Arksine/moonraker #544: Add support for A3dp slicer

robthide37, 2022-11-09:
### Is your feature request related to a problem? Please describe

Merrill is helping me make a new slicer that is built around idex and toolchangers etc. it is a superslicer fork .

### Describe the solution you'd like

Add an alias for A3dp slicer
 ; generated by A3dp-Slicer 2.5 on 2022-11-08 at 23:59:49 UTC

robthide37, 2022-11-15:
If i make the changes will you merge ?

Arksine, 2022-11-15:
I'm not opposed to adding support for forks, however I believe the addition should benefit a reasonable number of users.  Put differently, I don't think its appropriate to add aliases for everyone's personal fork.

I think the Klipper threshold of 100 users seems appropriate.  If so, I'll merge.

robthide37, 2022-11-25:
I have about 100 patrons and Merril is creating it for me and I have Daleguard working on getting rid of the iterative solver to support my servos. The slicer is going to be idex and toolchanger orientated so far it just has better variable layer height support as the main thing but next release is going to have granular pressure advance for each acceleration value and then prob non planar support but the main things are more designing proper toolchanger support and the ability to do things better like having a .4 nozzle do the perimeter and top but having a larger nozzle do the infill and supports with separate forced layer height for all vs right now you can set max layer height for support or sync but its very lacking.

robthide37, 2022-11-25:
Elaborating obv Daleguard is working on the kipper microcode lol and Merril the slicer . I read back and realized that was kinda confusing. I own Advanced3dprinting and the point is this is for others more than me on mine I just edited the aliases in metadata.py and it has it working but I want to have my users and anyone else which tbh is mainly troodon and modix users as well as LukesLabs stablebot users will be running idex and toolchangers . Anyway i want to grow more but need them to not have the teething pain of having me manually edit it or me have to make a fork and complicate things for them even more .

Arksine, 2022-11-25:
Sounds reasonable.  I'll merge a PR with an alias for your slicer if you want to submit it.

robthide37, 2022-11-26:
Ok ill ask merril to do it or if you want I can upload a gcode file "it just uses A3dp-Slicer then the release number and should follow the same output as supserslicer as of right now and I don't think that will be breaking anytime soon unless its for a really good feature . Ill pay you for your time since I greatly appreciate your work anyway so wouldn't mind contributing to you . I know in past your turned down offers from me but I have benefited and so have my Patrons and supporters so it is long overdue at this point .

Arksine, 2022-11-26:
Ok, I can add it, its a very small change and will only take a minute.  It would be good if you could upload a gcode file so I can do an integration test before merging.

I do have a Kofi link on the repo now if you want to contribute to Moonraker.  Thanks!.

robthide37, 2022-11-27:
[Calibration cube.gcode.txt](https://github.com/Arksine/moonraker/files/10098974/Calibration.cube.gcode.txt)
Here ya go

robthide37, 2022-11-27:
Sent 50 and il keep sending money periodically

Arksine, 2022-11-28:
Thanks for the contribution!  Support for A3dp has been added.  The metadata for the test file looks like the following:
```json
{
    "size": 6207668,
    "modified": 1669599717.170308,
    "uuid": "9cd0bbee-d334-496f-9fb5-e7dc92ebb834",
    "slicer": "A3dp-Slicer",
    "slicer_version": "2.5",
    "gcode_start_byte": 233865,
    "gcode_end_byte": 6192311,
    "layer_count": 249,
    "object_height": 40.38,
    "estimated_time": 4029,
    "nozzle_diameter": 0.4,
    "layer_height": 0.16,
    "first_layer_height": 0.3,
    "first_layer_extr_temp": 240,
    "first_layer_bed_temp": 110,
    "chamber_temp": 0,
    "filament_name": "3Dxtech CF ABS carbonx .4 nozzle RRF",
    "filament_type": "FLEX",
    "filament_total": 7997.52,
    "filament_weight_total": 24.81,
    "thumbnails": [
        {
            "width": 32,
            "height": 32,
            "size": 2130,
            "relative_path": ".thumbs/Calibration.cube-32x32.png"
        },
        {
            "width": 400,
            "height": 300,
            "size": 166119,
            "relative_path": ".thumbs/Calibration.cube-400x300.png"
        }
    ],
    "print_start_time": null,
    "job_id": null,
    "filename": "Calibration.cube.gcode"
}
```

robthide37, 2022-11-28:
Your a legend . If you need test hardware or anything please feel free to let me know . Me and Luke took over the huvud from bondus a few revisions back, I have my servos that I have Daleguard writing code for but are plug and play . Also I just made a CNC 9mmm upgrade kit for voron and troodons. Also I am partnered with Takoto so have tons of He20 and He50 and we just releases an idex kit for meter plus machines among other things.

robthide37, 2022-11-28:
Ill do a full reboot and more testing to make sure its not cached or anything . But it also it seems if I turn object processing on it breaks the preview and stuff .

robthide37, 2022-11-28:
Ok so ya something is not allowing it with object processing cause when I change to superslicer at the top of the file it instantly populates all the regex stuff and the thumbnail

Arksine, 2022-11-28:
Ah, the object processor does its own slicer identification (see the [preprocess-cancellation library](https://github.com/kageurufu/preprocess_cancellation).  It appears that there is an unhandled exception in the library when it is unable to identify the slicer.

I can handle this in `metadata.py`, however you may want the library to support your slicer natively as well.  It can be used to perform the post process directly in the slicer.

robthide37, 2022-11-28:
Ok so is that part of klipper ? If you can handle for now and just point me in direction and I will ask Kevin or Merril to help me depending on if I need to have code changed in the slicer or in klipper or both . Thaks again

kageurufu, 2022-11-28:
@robthide37 If your Superslicer fork is based on recent code, you could also include https://github.com/supermerill/SuperSlicer/pull/3208

Assuming your A3dp slicer is generating otherwise Slic3r-like Gcodes, it can just be added as another slic3r alias within https://github.com/kageurufu/preprocess_cancellation/blob/develop/preprocess_cancellation/slicers/__init__.py

Arksine, 2022-11-28:
Direct support in the slicer would certainly be ideal.