Arksine/moonraker #546: machine.shutdown function reboot system

Guilouz, 2022-11-10:
### What happened

machine.shutdown function reboot system and not shutdown

### Client

Mainsail

### Browser

Chrome

### How to reproduce

Just select Shutdown in Mainsail menu

### Additional information

OS : Ubuntu 20.04 LTS

Already tried sudo_fix.sh without success.

Also allow user 'pi' to reboot and shutdown without password without success.

systemd-logind service is already enabled is my distrib.

shutdown -H now command correctly shutdown my system but it seem machine.shutdown doesn't load this command.

Machine.reboot working as expected.

Arksine, 2022-11-11:
Please attach `moonraker.log` reproducing the issue as instructed.  By default Moonraker does not use the cli to shutdown, it uses logind.

Guilouz, 2022-11-14:
@Arksine Here the logfile : 
[moonraker.log](https://github.com/Arksine/moonraker/files/10007003/moonraker.log)

Arksine, 2022-11-15:
Looking at the log everything seems normal on Moonraker's end.   Its sounds like `systemd-logind` is not working correctly on your system if it reboot when shutdown is requested over DBus.

You could configure the cli fallback:

```
[machine]
provider: systemd_cli
```

The DBus provider is preferred, however this method should work if you have passwordless sudo.

Guilouz, 2022-11-15:
When I use systemd_cli nothing happens, shutdown do anything

Arksine, 2022-11-15:
I spun up an Ubuntu 20.04 VM and the commands work as expected.  There shouldn't be any need edit logind.conf.  I presume that this is a bug in the Ubuntu Distro specific to your device, or there is some other linux configuration causing this.

Unfortunately there isn't anything on my end that can fix your issue.  If you are running a Raspberry Pi, my recommendation is to install either Raspberry Pi OS or one of the Klipper specific alternatives.

Dark-Noir, 2023-01-16:
Hey, i found a "solution" for this. It happens on my FLSUN Speeder Pad (Pi), which runs on Ubuntu 20.04.5 LTS with kernel 4.9.191.

The Pad only shutsdown via Klipper Screen, not the Mainsail web UI.

I tried:
- sudo shutdown now
- sudo shutdown -P now
- sudo shutdown -h 0
- sudo shutdown -h -P now
- sudo poweroff

vie the SSH console without success.
I checked the Klipper Screen source and found "shutdown -H now" which works fine, via the console (even without sudo, the console throws an error, but it works).

So is there a way to let Mainsail/Moonraker to use the "sudo shutdown -H now" ?

Dark-Noir, 2023-01-16:
> It may be better to just eliminate the shorthand and use systemctl directly. Do the following commands work on your system?
> 
> ```
> sudo systemctl halt
> ```
> 
> Also reboot:
> 
> ```
> sudo systemctl reboot
> ```
> 
> FWIW, its still strange that the dbus provider does not work. It passes the request directly to systemd-logind.

Yeah both cmd's work fine here.

Arksine, 2023-01-19:
This should now be resolved with commit db81607370021925df8fec73539fa4f2c87276b1.

closed 2023-01-19 by Arksine

Arksine, 2023-02-08:
As a heads up there has been a change that is likely to affect users that reported this issue.  A new option has been introduced to the `[machine]` section named `shutdown_action`.  Its values can be `poweroff` or `halt`, with the default being `poweroff`.  Everyone here will likely need to add `shutdown_action: halt` to the `[machine]` section in their config.

The reason for this change is that some users desire the full `poweroff` action which "powers off" the machine after halting.  Since some distros don't support `poweroff` at all its necessary to make this an option.

Additionally the `halt` action should now work with the `systemd_dbus` provider.  That said, the halt action requires new polkit rules, so if you want to switch back to dbus you will first need to run `~/moonraker/scripts/set-policykit-rules.sh` to update the rules.