Arksine/moonraker #553: Add Basic Auth possibility

Disinterpreter, 2022-11-21:
### Is your feature request related to a problem? Please describe

`[web.py:log_exception()] - 401 GET /websocket (46.226.165.173): Basic Auth is not supported`

### Describe the solution you'd like

Add possibility to work with Basic Auth

### Describe alternatives you've considered

-

### Additional information

I just wanted to proxy my Mainsail to the internet, but with local Basic Auth (for security), but moonraker told me that it is impossible.

Thanks! Hope you get me.

EricZimmerman, 2022-11-21:
Related to this I would love to see the ability to set up an nginx reverse proxy which would solve this problem and open up a ton more flexibility. 

Most apps require a urlprefix of some kind for things to work right. I literally spent hours trying to get nginx working with rewrites and other things.

Example apps for reference include sonarr and radar. I can track down some more examples if needed but having it be some to put mailsail behind nginx allows for pretty much anything (ssl, different auth, etc)

Arksine, 2022-11-21:
Moonraker supports more robust authentication using JSON Web Tokens, however your front end needs to support it.  Fluidd does, I do not believe that Mainsail does.   Basic Auth is something that will not be supported in Moonraker itself, however as mentioned it should be possible to implement it using a reverse proxy.

Mainsail OS and FluiddPi both include an nginx configuration that handles the necessary rewrites.  I'm sure there is documentation available on getting Basic Auth configured for nginx.

EricZimmerman, 2022-11-21:
Ah this is news to me. If you mean this tho:

https://docs.mainsail.xyz/setup/manual-setup/mainsail

That is not what I am speaking of. 

I would like to be able to have another instance of nginx feed the web site being fed by the above configuration of website on the pi to the internet at large. 

I have a bunch of different internal services that I use a single external facing nginx proxy with authentication and I have not been able to make it work with mailsail. 

I will experiment however with the settings in that file above and see if I can't basically replicate nginx settings on the pi with my external facing instance

Arksine, 2022-11-21:
The configuration should be similar, just a difference in the address the proxy is listening on.   There may be additional changes required for an advanced configuration rewriting multiple services on the same proxy.

One thing I would add more general to the topic...Basic Auth is fine for protecting local instances from children, siblings, roommates, etc.  It isnt sufficient for protecting an instance exposed to the WAN.  This is the primary reason why Moonraker doesn't have native support.  Its better than nothing, but that isn't saying much.

Arksine, 2022-11-21:
Indeed, it is my opinion.  If you believe that Basic Auth is sufficient for your needs then it is certainly your decision to use it.