Arksine/moonraker #556: Large file upload fails when object processing is enabled

vavaroutsos, 2022-11-25:
### What happened

When uploading a large file with object processing enabled, the file upload fails on the client side. 100% of the file is transmitted, but no reply is posted until after object processing completes. On a Pi 3B this was taking over a minute and the client side appeared to be timing out. I think once 100% of the file is received, the response should be posted and then object processing should be done.

I also saw an information entry in the moonraker log about shapely not being found and complex hulls being disabled. I installed shapely for both python 2 and 3, but the problem still persists. However, this doesn't seem to be related to the root cause of the file upload issue.

### Client

Mainsail, Other

### Browser

Chrome, Other or N/A

### How to reproduce

Upload a very large gcode file with object processing enabled. I tested using the Cura plugin and Mainsail/chrome. Both failed with similar entries in the moonraker log.

### Additional information

[moonraker.log](https://github.com/Arksine/moonraker/files/10094989/moonraker.log)

Arksine, 2022-11-25:
Moonraker is returning 201 on all of the upload requests in your log, it doesn't timeout requests.  Your issue is likely with your nginx configuration.  In Cura you can try bypassing the proxy by adding the port to the url (ie: `http://my-printer:7125`) in the upload configuration.

There are some [nginx configuration options](https://nginx.org/en/docs/http/ngx_http_proxy_module.html) that control this behavior. Specifically two options that increase the default read and send timeouts (which are 60 seconds):
```
proxy_read_timeout 1d;
proxy_send_timeout 1d;
```
On MainsailOS the nginx configuration is at `/etc/nginx/sites-available/mainsail`.   Put the above options in the `server` block, then restart the nginx service.   One thing to keep in mind is that you need root access to modify the file, so for example if you use nano:
```
sudo nano /etc/nginx/sites-available/mainsail
```

Moonraker holds the request until metadata and postprocessing are done, there are several technical reasons for this.  One is so front ends can be sure metadata is available when the upload returns, they like to immediately present this info to users.  With respect to object processing, we wouldn't want to upload the file and return immediately, as the user may choose to start the print before object processing is complete.

vavaroutsos, 2022-11-26:
Eric, thanks for the detailed reply. These changes corrected the problem.

 

Best Regards,

 

Pete

Arksine, 2022-11-28:
Good to hear, thanks.

closed 2022-11-28 by Arksine