Arksine/moonraker #558: Klipper 0.11 breaks "exclude object" support.

bsas, 2022-11-29:
### What happened

Looks like after I upgraded my klipper to 0.11 the "exclude object" section disappeared.

### Client

Mainsail

### Browser

Chrome

### How to reproduce

Upgrade to klipper 0.11. Even with [exclude_object] in my printer.cfg and "enable_object_processing: True", the exclude object function doesn't show up. It was working before on 0.10.xxx

Arksine, 2022-11-30:
If you are looking for assistance with Klipper you may have better luck on the [discourse](https://klipper.discourse.group/categories).  I'm unsure as to why 0.11 would break exclude object.  

That said, a recent change was made to object processing in Moonraker.  If you provide `moonraker.log` reproducing the issue (ie: upload a file that requires object processing then download the log) then I would be willing to look at it and make sure object processing is behaving as it should in Moonraker.

bsas, 2022-12-01:
OK, ignore me, my mistake. I changed computers and the slicer on my computer wasn't configure to generate label per object :P

closed 2022-12-01 by bsas