Arksine/moonraker #567: Cannot install new moonraker

ymanyin, 2022-12-15:
### What happened

Cannot install Moonraker.
Failed  on kiauh installation.
Then I tried manually install by:
cd ~/moonraker/scripts
./install-moonraker.sh

### Client

Mainsail

### Browser

Chrome

### How to reproduce

Reproduce every time runs
cd ~/moonraker/scripts
./install-moonraker.sh

### Additional information

###### Installing python virtual environment...
Collecting tornado==6.2.0
  Using cached tornado-6.2.tar.gz (504 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting pyserial==3.4
  Using cached pyserial-3.4-py2.py3-none-any.whl (193 kB)
Collecting pyserial-asyncio==0.6
  Using cached pyserial_asyncio-0.6-py3-none-any.whl (7.6 kB)
Collecting pillow==9.3.0
  Downloading Pillow-9.3.0.tar.gz (50.4 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╸ 50.2/50.4 MB 4.1 MB/s eta 0:00:01./install-moonraker.sh: line 49:  1978 Killed                  ${PYTHONDIR}/bin/pip install -r ${SRCDIR}/scripts/moonraker-requirements.txt

ymanyin, 2022-12-15:
Tried all day, it always stop at downloading Pillow-9.3.0

I also tried to manually install Pillow-9.3.0. It failed every time.
What can I do to resolve the situation? Thanks in advance.

  Downloading Pillow-9.3.0.tar.gz (50.4 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╸ 50.3/50.4 MB 4.1 MB/s eta 0:00:01Killed

Arksine, 2023-01-19:
Unfortunately this appears to be some kind of OS issue and there is little I can do.  Hopefully in the time since you created this you were able to resolve it with assistance from the discord or discourse.  Perhaps an update to Armbian is necessary to fix the issue.

closed 2023-03-07 by Arksine