Arksine/moonraker #568: MoonRaker wont connect to Klipper

Drkmttr82, 2022-12-17:
### What happened

Set up main raspberry pi with the mainsail img. then used kiauh to install the rest. Able to ssh into pi and upload firmware to printer. getting this error when it comes to moonraker connecting to klipper. 

{"error": {"code": 404, "message": "Not Found", "traceback": "Traceback (most recent call last):\n\n  File \"/home/HammaForge/moonraker-env/lib/python3.9/site-packages/tornado/web.py\", line 1713, in _execute\n    result = await result\n\n  File \"/home/HammaForge/moonraker/moonraker/app.py\", line 737, in get\n    self.absolute_path = self.validate_absolute_path(\n\n  File \"/home/HammaForge/moonraker-env/lib/python3.9/site-packages/tornado/web.py\", line 2813, in validate_absolute_path\n    raise HTTPError(404)\n\ntornado.web.HTTPError: HTTP 404: Not Found\n"}}

I checked the config file under trusted clients and my IP is in there. So im unsure of what is the issue. Ill upload the moonraker log as well. Thanks in advance for any help! 
[moonraker.log](https://github.com/Arksine/moonraker/files/10250877/moonraker.log)

### Client

Mainsail

### Browser

Chrome

### How to reproduce

Wont connect on port 7125 or the trusted clients li
[moonraker.log](https://github.com/Arksine/moonraker/files/10250879/moonraker.log)
st.

Patbonamy, 2022-12-19:
Same for me with KlipperScreen.
Please help!!!
Thanks

Arksine, 2023-01-19:
I hope that you were able to find assistance for this in the discord or discourse.  That said, I don't see any 404s in the log, my guess is that there is an issue with Klipper preventing it from starting.  Moonraker cannot connect to the unix socket until Klipper creates it.

Arksine, 2023-07-03:
Discussion appears to have concluded on this issue, so I am closing it.

closed 2023-07-03 by Arksine