Arksine/moonraker #572: Support for multiple bound_service in [power] configuration

cmerkle, 2022-12-21:
### Is your feature request related to a problem? Please describe

Moonraker currently doesn’t seem to support multiple bound services for [power device_name]. It would be really nice to be able to bind multiple services to the power of a single device, for example, bind klipper and webcamd to my printer’s power so both the klipper service and my webcam shutdown when the printer is powered off. 

### Describe the solution you'd like

Implement support for multiple bound services for power devices, either separated by commas or new line characters:

[power my_printer]
bound_service: klipper, webcamd

OR

[power my_printer]
bound_service: 
    klipper
    webcamd

pinkpawkitty, 2022-12-26:
I have a fork that implements that.

@Arksine Is this a feature you would consider merging, aka shall I create a PR?

cmerkle, 2022-12-27:
> I have a fork that implements that.
> 
> 
> 
> @Arksine Is this a feature you would consider merging, aka shall I create a PR?

Would love to see this merged!!

pinkpawkitty, 2023-01-04:
@cmerkle seems to be fixed in https://github.com/Arksine/moonraker/commit/b3a9447392b758dc8270ef59052282e8677f1500 so I think you can close this issue

Arksine, 2023-01-09:
Yes, I added support for this in the linked commit.  I had previously implemented this but held off on merging until I came up with a safe way for users add additional services to the allow list.

closed 2023-01-09 by Arksine