Arksine/moonraker #576: Filament database reboot?

Digityle, 2022-12-25:
### Is your feature request related to a problem? Please describe

Apparently the filament database/usage tracker @mateuszbrzezinski was working on has been completely abandoned, I've seen many posts on reddit requesting something like this.

### Describe the solution you'd like

Would love the ability to have a component that lists what filaments/colors I have in stock and amount left on spools. I have an hx711 strain gauge scale and rfid tagging for switching rolls (ran off esp32 and homeassistant) 

### Describe alternatives you've considered

I'm currently running mainsail in an iframe inside home assistant, worst case scenario I'm willing to run into the wall repeatedly until I learn enough python to cobble it together myself lol.

alexandrkut, 2023-02-28:
I made a hook that pulls the url with information about the spent filament after printing.
And then you can do whatever you want with this information (I have a terrible, but a filament accounting system that suits me).

Arksine, 2023-07-09:
Filament tracking in now supported via `Spoolman`.  Thanks.

closed 2023-07-09 by Arksine