Arksine/moonraker #579: [MQTT] publish_split_status is ignored

mplogas, 2023-01-01:
### What happened

According to the documentation, setting 
```
[mqtt] publish_split_status: True
```
should result in hierarchical MQTT topics, e.g. ```myprintername/klipper/state/extruder/temperature```

With Moonraker v0.7.1-790-g358aee9 and Mosquitto v2.0.14 this setting does nothing, resulting in a single JSON-style status

```{"eventtime": 171289.731687144, "status": {"toolhead": {"estimated_print_time": 6621.401510464286}, "heater_bed": {"temperature": 23.53}}}```

config:
```
[mqtt]
address: -redacted-
port: 1883
username: {secrets.mqtt.username}
password: {secrets.mqtt.password}
status_objects:
    toolhead
    extruder
    heater_bed
    print_stats
    display_status
    virtual_sdcard
publish_split_status: True
```

### Client

Other

### Browser

Other or N/A

### How to reproduce

1. set up mqtt broker
2. set up moonraker secrets
3. set up moonraker mqtt client in accordance with documentation
4. restart moonraker
5. use any mqtt client to monitor the myprintername/klipper topic (e.g. mqtt explorer)

### Additional information

[moonraker.log](https://github.com/Arksine/moonraker/files/10329468/moonraker.log)

Arksine, 2023-01-01:
This is because support for "split status" was added in commit e3bcfb5ea0a3256bdbe19855d20c4431b57cd398, which is the commit after the version of Moonraker you are currently running.

mplogas, 2023-01-01:
Good catch, thank you! 
Mainsail is not reporting a Moonraker update, so I didn't bother to check via KIAUH. I updated Moonraker to  v0.7.1-791 and now the feature works as expected.

closed 2023-01-01 by mplogas