Arksine/moonraker #585: E: Unable to locate package python3-libgpiod on AMD64 ubuntu 18.04

anand-010, 2023-01-11:
### What happened

I have updated python from 3.6 to 3.10 after i've tried to install moonracker and i've got error E: Unable to locate package python3-libgpiod.

### Client

Mainsail, Fluidd

### Browser

Chrome

### How to reproduce

Try to install on non arm debian.

Arksine, 2023-01-19:
Ubuntu 18.04 is likely too old to install Moonraker via the default install script, as its intended for Debian 9 and above (Ubuntu 20.04 should work).  You will likely need to create your own install script, you should be able to remove the libgpiod dependency if you don't intend to use the `power` component.

manic101, 2023-02-10:
Having the same problem with Ubuntu 18.04 
I am running it in an old laptop that is 32 bit, so need to use this version of ubuntu

Arksine, 2023-02-10:
[Debian Bullseye](https://www.debian.org/releases/bullseye/debian-installer/) has an i386 installer.  As mentioned, Ubuntu 18.04 ships without dependencies Moonraker requires.  You are free to try to install on that machine, however it is likely going to be nothing but trouble and is unsupported.

Arksine, 2023-06-11:
As this discussion seems to have concluded I am going to close this.

closed 2023-06-11 by Arksine