Arksine/moonraker #587: Update Manager should work with repos without tags

GaryBoone, 2023-01-17:
### Is your feature request related to a problem? Please describe

Moonraker's update manager can't work with some GitHub repos because it explicitly [requires semantic version tags](https://github.com/Arksine/moonraker/blob/master/docs/configuration.md#all-other-extensions). Some github repos don't have tags, yet contain important codebases for 3D printers. Two popular examples are:

- `Klicky`, found in https://github.com/jlas1/Klicky-Probe
- `stealthburner_leds.cfg`, found in https://github.com/VoronDesign/Voron-Stealthburner/

Moonraker should include a way to keep these repos updated on our printers. 

### Describe the solution you'd like

Ideally, the user would be able to specify what kind of updates they want, including `all`, `major`, `minor`, and `patch`.

### Describe alternatives you've considered

One solution could be to not require semantic tags, perhaps optionally. 

Another would be to add on option like `include_all_updates` that defaults to false, but when true updates the local repo with each repo commit.

### Additional information

This feature would also allow users who want the latest updates to receive them without waiting for a new tag.

As there is already a `refresh_interval` parameter, overly pinging GitHub with update checks should not be an issue.

User Expectations:
- Users expect updates from GitHub to work with all GitHub repos. Limiting to just tagged ones is surprising.
- Users who request an update want receive them without waiting for a new tag. Seeing nothing update even though the repo says, `a day ago` on a file, for example, is surprising.
- Users expect an update from GitHub to return the current version on GitHub. Limiting to semantic versions violates expectations without some explicit flag setting to require semantic versioning.

Frix-x, 2023-01-23:
I fully agree with that : if there is no semantic versionning, we could just default to update on every commits ?

Also on the same subject, I think it should be possible to support github repository that use `main` and not `master` as the main branch name. Maybe specifying the "target" branch in the moonraker update manager config could be a solution ?

Arksine, 2023-02-07:
I think you have misinterpreted the note.  Moonraker requires a tag to identify the version, not to update the repo.  Moonraker will happily update a repo without a tag as long as the repo is configured on the `dev` channel.   That said, Moonraker reports `?`  for the version as it cannot extract the version info from the output of `git describe`.  Front ends disable update controls without such info.

> Also on the same subject, I think it should be possible to support github repository that use main and not master as the main branch name. Maybe specifying the "target" branch in the moonraker update manager config could be a solution ?

This is already supported by the `primary_branch` option.

Frix-x, 2023-02-08:
> This is already supported by the `primary_branch` option.

Yes thanks, I discover it in the meantime... Sorry

> I think you have misinterpreted the note. Moonraker requires a tag to identify the version, not to update the repo. Moonraker will happily update a repo without a tag as long as the repo is configured on the `dev` channel. That said, Moonraker reports `?` for the version as it cannot extract the version info from the output of `git describe` without a tag. Front ends disable update controls without such info.

This is useful information and I did not understood the documentation like that. Maybe adding this precision to it could help others as well ? (keep in mind that I'm not a native english speaker so it could be me)

Arksine, 2023-02-08:
Technically it would be an issue with the front end.  The question is if they should enable update controls when the commit hashes are different.  Moonraker reports those as well as the versions extracted from the tags.

With regard to the documentation, it probably doesn't hurt to add precision.

GaryBoone, 2023-02-20:
Given the above, here's a Moonraker upate manager config that works for `stealthburner_leds`. 

```
# Update stealthburner_leds
# The stealthburner_leds is found in the Voron-Stealthburner repo:
# https://github.com/VoronDesign/Voron-Stealthburner/tree/main/Firmware/stealthburner_leds.cfg
[update_manager stealthburner_leds]
type: git_repo
channel: dev
path: ~/Voron-Stealthburner
origin: https://github.com/VoronDesign/Voron-Stealthburner.git
primary_branch: main
is_system_service: False
```
I see that the [moonraker documentation](https://github.com/Arksine/moonraker/blob/master/docs/configuration.md#git-repo-configuration) has been updated to clarify. 

When there was an update, it looked like this:
[image]

Now that there is no update, it looks like this:
[image]

That doesn't quite seem right, as it is not unknown, it is updated to the latest version in the repo. All that's unknown is the semantic version.

But as noted ab

closed 2023-02-20 by GaryBoone