Arksine/moonraker #588: Unable to generate file metadata

tinntbg, 2023-01-17:
### What happened

Hello, I'm using PrusaSlicer to directly upload gcode files to my Fluidd Voron v0. Everything was working correctly until the last update. Currently it's failing to generate the file matedata. 

In the logs I'm seeing the following error: 
`2023-01-17 15:17:08,438 [shell_command.py:_check_proc_success()] - Command (/home/pi/moonraker-env/bin/python /home/pi/moonraker/moonraker/components/file_manager/metadata.py -p /home/pi/gcode_files -f "ABS_0.3mm_piller - 19.3156g - 52m.gcode") successfully finished
2023-01-17 15:17:08,438 [file_manager.py:_process_metadata_update()] - Error running extract_metadata.py
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/components/file_manager/file_manager.py", line 1959, in _process_metadata_update
    await self._run_extract_metadata(fname, ufp_path)
  File "/home/pi/moonraker/moonraker/components/file_manager/file_manager.py", line 2001, in _run_extract_metadata
    decoded_resp: Dict[str, Any] = json.loads(result.strip())
  File "/usr/lib/python3.7/json/__init__.py", line 348, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.7/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.7/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
2023-01-17 15:17:08,439 [file_manager.py:_process_metadata_update()] - Unable to extract medatadata from file: ABS_0.3mm_piller - 19.3156g - 52m.gcode`

I have attached the full log file:
[moonraker.log](https://github.com/Arksine/moonraker/files/10436530/moonraker.log)

Can you please assist me with this issue?

Thanks!

### Client

Fluidd

### Browser

Chrome

### How to reproduce

Generate a file with PrusaSlicer and use the automatic upload of the file.

Arksine, 2023-01-17:
I think you likely have a corrupt moonraker repo.  You can ssh into your Pi and try the following:
```
cd ~
sudo service moonraker stop
rm -rf moonraker
git clone https://github.com/Arksine/moonraker.git
sudo service moonraker start
```

Removing power from the Pi/SBC without safely shutting down is one of the primary causes of repo corruption.  Something to keep in mind if your Pi is on the same power supply as your printer.

tinntbg, 2023-01-18:
Thanks @Arksine that solved my issue. I have multiple printers running the same way and I haven't had an issue until now. I'll keep this in mind thanks again!

closed 2023-01-18 by tinntbg