Arksine/moonraker #589: 'Error sending homeassistant command: info' when everything is correct

SoundedSquash, 2023-01-18:
### What happened

I get the following error when starting up moonraker and the switch is grayed out and I can't click it.

> 2023-01-18 16:45:47,900 [klippy_connection.py:_do_connect()] - Klippy Connection Established
2023-01-18 16:45:47,957 [power.py:init_state()] - Device Init Error: ender_3
Error sending homeassistant command: info

Here's a screenshot of the config
[image]

Looking at the code for power.homeassistant, everything looks right. It goes down the info path on 1121, but errors on line 1138.
[image]

The URL translates to GET http://pi.local:8123/api/states/switch.ender_3 and that works fine in Postman
[image]

The only thing I can think of is the token. Line 1109 uses config.gettemplate instead of config.get. I don't see documentation on what that could mean or how my moonraker.conf should look. It would also be nice to see the error code/message from the response logged, and not just the command that failed.

### Client

Fluidd

### Browser

Chrome

### How to reproduce

Set up a home assistant config and restart moonraker to load it.

Arksine, 2023-01-19:
I agree that the traceback should be logged during initialization and pushed 2ad7a6a0871d4f346d94853db459e539701da881 which makes that change.

While there is no developer documentation for `gettemplate()` at the moment, the [configuration documentation](https://moonraker.readthedocs.io/en/latest/configuration/#home-assistant-configuration-http) does indicate that the option accepts templates and directs the user to read the [secrets documentation](https://moonraker.readthedocs.io/en/latest/configuration/#secrets).  In this particular case the token can be placed in a secrets file and the option can render a jinja2 template to access it.  This way it isn't necessary to put credentials in the configuration.  That said it will render fine when you add the token directly to the configuration.

At the moment I would recommend pulling the new commit to see if the traceback provides an additional clue as to what the issue is.  Hopefully you will be able to resolve it, otherwise attach the full `moonraker.log` and I will attempt to take a look.

SoundedSquash, 2023-01-19:
@Arksine Thanks for the quick updates! I applied the commit and noticed that it gave an error `socket.gaierror: [Errno -2] Name or service not known`. Google search led me to it not resolving the address. Turns out, I can't resolve pi.local on this other machine (Orange Pi 5), but it works on Windows.
[moonraker.log](https://github.com/Arksine/moonraker/files/10452602/moonraker.log)

Anyway, I updated the address to the local IP and it worked! 

As for secrets, I'm still having trouble. I have two power entries to debug some issues. Ender_3 works fine when I put in the token manually. Denroom does not pull in the token and fails. Also, it warns that "domain" is not being used? Anyway, attached is the latest restart of the [moonraker.log](https://github.com/Arksine/moonraker/files/10452602/moonraker.log). It's saying there is no entry for home_assistant in the secrets file.
[moonraker.log](https://github.com/Arksine/moonraker/files/10452602/moonraker.log)

I also believe I have the secrets setup correctly. It's all in the same folder.
[image]

Arksine, 2023-01-19:
The `domain` warning is due to a cascading failure from the inability to parse the `token` option, once you have secrets properly set up it will resolve itself.

The `moonraker.secrets` file has to be in the data path, on your machine it should be located at `/home/pi/printer_data/moonraker.secrets`.   The purpose of this file is to protect credentials from network exposure, so you will not be able to access it from Fluidd.

You will need to use `ssh` to complete the setup, moving the file would probable suffice.  It should be currently located at `/home/pi/printer_data/config/moonraker.secrets`.

SoundedSquash, 2023-01-19:
That was it, wrong location. Works good now.

Thanks for all your work.

closed 2023-01-19 by SoundedSquash