Arksine/moonraker #594: klippy.log empty

legend069, 2023-01-22:
### What happened

using the fluidd web GUI klippy.log is empty, been like that for a while when ever i tried to look at it.

loaded up winscp to take a look and noticed something funny has happened to the log numbering.

[image]

### Client

Fluidd

### Browser

Firefox

### How to reproduce

several rebooting, and still don't get a log file.

if i open klippy.log1 it looks like it should and has a correct time/date stamp with it.

guess it all started form https://github.com/Arksine/moonraker/issues/516

Arksine, 2023-01-23:
Moonraker itself should not modify the file names.  Without `moonraker.log` I can't diagnose this, however from the screenshot it appears that some other software is modifying the log files and possibly compressing them.

legend069, 2023-01-25:
sorry, thought i uploaded the logfile.
[moonraker(12).log](https://github.com/Arksine/moonraker/files/10495447/moonraker.12.log)

the only other software i think i have installed i samba,
otherwise it's just the standard klipper/fluidd stuff.

Arksine, 2023-01-25:
There isn't anything in the log that stands out other than this:
```
Unsafe Shutdown Count: 273
```
When you remove power from the Pi without safely shutting down the OS then strange things can happen, usually as a result of file system corruption.   That said, something on your machine is renaming the log files (see the `klippy.log.1`, `moonraker.log.1`).  There are also compressed files (see the `.xz` files).  If you are using a specific distrubution (such as FluiddPi) then you might want to see assistance in the Fluidd Discord, GitHub, etc.

legend069, 2023-01-25:
that doesn't explain how at the moment klipper is currently writing logs to the wrong file though?

legend069, 2023-02-03:
> 

i'll close this issue now, i've wiped the card and started again.

closed 2023-02-03 by legend069