Arksine/moonraker #598: Uploading large gcodes results in "504 Gateway Timeout"

Arakon, 2023-01-28:
### What happened

Whenever I upload a larger file (guessing >10 or 20 MB) from Superslicer, Bambu Studio or Cura directly, it uploads fine, file is complete, prints fine, etc, but after a minute or so, the slicer pops up with "504 gateway timeout". Also seen it happen when uploading through the Fluidd web interface. It shows "Processing" for 30 seconds or so, then throws the 504 popup.
[image]
[image]

This is happening on two independant printers, from two seperate PCs, both on the latest versions on the stable channel. Any ideas?
A month or two back this didn't happen.

### Client

Fluidd

### Browser

Firefox, Other or N/A

### How to reproduce

- Generate a large gcode file
- Upload it by the way of your choice

### Additional information

[moonraker.log](https://github.com/Arksine/moonraker/files/10528037/moonraker.log)

Arksine, 2023-01-28:
This looks like a duplicate of #556, nginx is causing the timeout.  As a test you can connect directly to Moonraker by including the port with the hostname/ip (ie: `http://your-hostname.local:7125`).   The linked issue has instructions on how to correct the problem on MainsailOS, I suspect the procedure would be very similar in FluiddPi.  The name of the file in `sites-available` is probably `fluidd` rather than `mainsail`.

Arakon, 2023-01-28:
Indeed, that fixed it. I had searched for the "gateway timeout" error I got, so I didn't find that issue.

closed 2023-01-28 by Arakon