Arksine/moonraker #599: API call for object query no longer reports total_layers and current_layer from Mainsail

davisgoodman, 2023-01-29:
### What happened

I have an API call which feeds my Home Assistant 3D printer status page. Not sure if it's following an update but I no longer get the Total_layer count nor the current_layer from my API call:

http://<ip_address>:7125/printer/objects/query?print_stats&display_status  returns the following:

{"result": {"eventtime": 761501.038450343, "status": {"print_stats": {"filename": "Voron/LDO_Mods-Anycubic Mega X-Sparta 3D ABS+.gcode", "total_duration": 8517.018012062996, "print_duration": 7014.033033833955, "filament_used": 10986.244910000241, "state": "printing", "message": "", "info": {"total_layer": **null**, "**current_layer**": null}}, "display_status": {"progress": 0.17, "message": "Printing..."}}}}

### Client

Other

### Browser

Chrome

### How to reproduce

I get it just by doing the a regular http call through the browser as shown above

### Additional information

Moonraker version v0.7.1-809
Mainsail 2.4.1

Arksine, 2023-01-29:
Hi.  Presumably your issue is that the G-Code file you are printing does not invoke the [SET_PRINT_STATS_INFO](https://www.klipper3d.org/G-Codes.html#set_print_stats_info) command.

davisgoodman, 2023-01-29:
> Hi. Presumably your issue is that the G-Code file you are printing does not invoke the [SET_PRINT_STATS_INFO](https://www.klipper3d.org/G-Codes.html#set_print_stats_info) command.

Ok I'll check this but, I've never inserted the command in any of my Gcodes and it worked in a few cases. Also the thumbnails doesn't get updated. I should have mention it in the original post.

Arksine, 2023-01-29:
It isn't possible to set those values without calling that gcode AFAIK.  Regardless this information is produced and reported Klipper, Moonraker simply passes the request through.

With regard to thumbnails, you'll need to clarify.  If you can reproduce a situation where you upload a gcode file that contains  valid thumbnails and they are not extracted in the `thumbs` directory relative to the gcode file I'm willing to look at it.  If the thumbnails are there then the issue doesn't likely lie with Moonraker.

davisgoodman, 2023-01-29:
Strangely enough though, in Mainsail I do have all the data show up in the WebUI
[image]
I've added the [print_stats] section in mainsail.cfg but currently have a 10h print job running. Will test after.

Arksine, 2023-01-29:
Mainsail has a fallback procedure that can estimate the total layer height and current layer when they aren't present in the API call.  These estimates are not always accurate, particularly on objects with supports and/or adaptive layer.  This is why the `SET_PRINT_STATS_INFO` gcode command was added, if you want that data to show up in API calls you will need to make the correct adjustments to your slicer.

Adding `[print_stats]` to the configuration isn't necessary, its automatically loaded by `[virtual_sdcard]`.

davisgoodman, 2023-01-29:
Ok thanks. Totally makes sense as in many occasions I didn't feel the displayed current layer was accurate in many occasions

davisgoodman, 2023-01-29:
> It isn't possible to set those values without calling that gcode AFAIK. Regardless this information is produced and reported Klipper, Moonraker simply passes the request through.
>
> With regard to [...]

Ok so, as far as the thumbnails, I can confirm that none of the gcode files have corresponding thumbnails in the .thumbs directory relative to the gcode which may explain why it doesn't come up in the api call. 

Are there special permissions required to write in the directory?

Arksine, 2023-01-29:
Most likely Mainsail has implemented a filter to prevent users from viewing the thumbnail files.  The API will return a list of the thumbnail files when the contents of a directory are requested.

davisgoodman, 2023-01-29:
Ok will look into it on the Mainsail side. 

Once my print finishes I will be able to implement the status command in my next prints and will update.

 

Thanks for the swift reply

davisgoodman, 2023-01-30:
> Most likely Mainsail has implemented a filter to prevent users from viewing the thumbnail files. The API will return a list of the thumbnail files when the contents of a directory are requested.

Hi, after further testing, I believe the issue of the thumbnail not displaying is that the relative path given in the metadata object query doesn't include the subdirectory so when using the path it only gives .thumbs/file.png where it should actually be subfolder/.thumbs/file.png

unless there it's possible to subfolder variable.

Arksine, 2023-01-30:
The `relative_path` is relative to the gcode file's parent folder, this is nearly always going to be `.thumbs/<file_name>-<pic_size>.png`.

If the frontend is requesting a thumbnail from the wrong location there will be `404` errors in the log.

davisgoodman, 2023-01-30:
Ok but Mainsail allows creating subfolders. For a particular project I created a folder called Voron where I uploaded all the files to be printed in that subfolder.

In the metadata the filename is Voron/file.gcode but the thumbnail relative path is given as .thumbs/file.png and not Voron/.thumbs/file.png

Arksine, 2023-01-30:
The thumbnail path is *relative* to the the gcode file's parent.  Therefore, if the gcode file is in a `Voron` subfolder, it is understood that the `.thumbs` is also in the `Voron` subfolder.  The front ends are aware of this and will join the gcode file's parent to the location specified in the `relative_path` field.  And as I mentioned previously if the front is requesting the wrong location there will be `404` errors in the log.

davisgoodman, 2023-01-30:
As mentioned before that is something that the HomeAssistant API call will have to adjust. As I'm not a developer as such I'll pass this on. Is there an api call to get the parent directory only?

davisgoodman, 2023-01-30:
> Ok thanks. Totally makes sense as in many occasions I didn't feel the displayed current layer was accurate in many occasions
> […](#)

Now the layer count works. 

I found another issue but let me know if you rather have me open another ticket.

If the gcode filename includes a "+" sign, the API call fails. The file output of the slicer includes the type of filament which in this case is ABS+

Arksine, 2023-01-30:
> As mentioned before that is something that the HomeAssistant API call will have to adjust. As I'm not a developer as such I'll pass this on. Is there an api call to get the parent directory only?

Understood.  There is no API to get the parent directory, however it is reported as part of the `filename` field in the `print_stats` object.  Presumably Home Assistant provides a way for you to modify the response, such as through a Jinja2 template. it is a limitation that they need to address.

> If the gcode filename includes a "+" sign, the API call fails. The file output of the slicer includes the type of filament which in this case is ABS+

This is not an issue with Moonaker, you need to url escape the filename if you are placing it in the http path or query string.

davisgoodman, 2023-01-30:
> > As mentioned before that is something that the HomeAssistant API call will have to adjust. As I'm not a developer as such I'll pass this on. Is there an api call to get the parent directory only?
> 
> Understood. There is no API to get the parent directory, however it is reported as part of the `filename` field in the `print_stats` object. Presumably Home Assistant provides a way for you to modify the response, such as through a Jinja2 template. If not it is a limitation that they need to address.
> 
> > If the gcode filename includes a "+" sign, the API call fails. The file output of the slicer includes the type of filament which in this case is ABS+
> 
> This is not an issue with Moonaker, you need to url escape the filename if you are placing it in the http path or query string

closed 2023-01-30 by davisgoodman