Arksine/moonraker #602: Weird interaction with klipperscreen pkglist on bullseye + python3

miklschmidt, 2023-01-31:
### What happened

Issue described here: https://github.com/jordanruthe/KlipperScreen/issues/883

I'm unsure whether this is a moonraker issue or a KlipperScreen issue or something else entirely.

### Client

KlipperScreen

### Browser

Other or N/A

### How to reproduce

1) Install https://github.com/Rat-OS/RatOS/releases/tag/v2.0.0-beta1 i would strongly assume it can be reproduced with MainsailOS 1.0 as well (In that case you would need to install KlipperScreen at a commit before the one that changed the install script.)
2) Update KlipperScreen through mainsail / moonraker.
3) Tons of seemingly unrelated packages missing, if you restart any of the services they'll fail. Rebooting leave you with the majority of the services dead (Klipper, Moonraker, RatOS-configurator, etc etc).

Arksine, 2023-01-31:
I couldn't reproduce this on vanilla RPi OS Bullseye.  I'll try to reproduce it with your distro when I get more time.  In the meantime, if you have a chance it would be helpful if you could repro this with debug logging enabled (add `-v` to the MOONRAKER_ARGS in `~/printer_data/systemd/moonraker.env`) then attach the resulting `moonraker.log`.

I suspect that this is something strange going on with the system.  Moonraker never requests that PackageKit remove a package, therefore it would seem that PackageKit is taking this action on its own accord.  Needless to say that this isn't ideal.  I may need to default to the APT CLI implementation on systems that support it until we can determine what is causing this.

miklschmidt, 2023-01-31:
@Arksine You're the best! Thank you for the prompt action! I'm not even off work yet myself, let me know if there's anything i can do to help!

Arksine, 2023-01-31:
I have pushed the changes.  I wiped the RatOS image and tested these changes again to be sure and the update completed successfully.  I'll wait for your feedback on testing a new build before I close this.  I assume its possible to reset the KlipperScreen repo to a commit prior to the script change in a new build, however I haven't familiarized myself with how CustomPiOS works.

I have another change I plan to make which will improve this a bit more, however I think its best to be absolutely sure the issues with PackageKit are resolved before I push them.

miklschmidt, 2023-01-31:
Confirmed! It works :)

miklschmidt, 2023-02-01:
@Arksine should i close this or do you want to push the other changes first, so we can verify those as well?

Arksine, 2023-02-01:
We can close this since it works on your system.  The upcoming changes I can reliably test locally, basically just an improvement on package detection.  Moonraker will parse system packages and/or python requirements before and after an update.  Only new packages/modules detected in the files will be installed.  This way we avoid triggering PackageKit and/or Pip when one of the files are changed without a new dependency.

Additionally I intend to reintroduce pip upgrades.  This time I will perform a version check before the upgrade and use a pinned version.

closed 2023-02-02 by Arksine