Arksine/moonraker #605: Moonraker thinks it needs a sudo password - I strongly disagree

sboschert, 2023-02-02:
### What happened

I just upgraded Moonraker from a half year old version to the latest one. Now, on every Moonraker restart I am greeted with the following notification:

"Sudo Password Required
Moonraker requires sudo access to finish updating. Please click on the attached link and follow the instructions."

I am not using Moonraker's install script because I have an automatic way of setting up klipper machines via ansible. However, I am doing pretty much the same things it does - just via ansible. Now I realize that this is not the way Moonraker was intended to be installed. However, Moonraker is working fine and I see no reason why it should need a sudo password.

Is there any way to disable this notification in a way that it doesn't re-appear after a Moonraker restart? Alternatively, could you please let me know what is triggering this warning so that I can replicate a possible fix that will satisfy Moonraker via ansible?

### Client

Mainsail

### Browser

Firefox

### How to reproduce

see above. Basically: Do a custom install of Moonraker. ;)

Arksine, 2023-02-02:
Moonraker needs to update the systemd unit file to accommodate a file structure change merged in PR #491.   More details are in the announcement, issue #516.

You are free to install Moonraker however you feel best, it isn't apparent to me why you would use the `[update_manager]` with Ansible.  Presumably it is possible to use Ansible itself to update your installation.  With regard to the warning, you can eliminate it by updating the service file to match the required file.  You can provide Moonraker sudo access to automate it or you can do it manually.

sboschert, 2023-02-02:
Thanks for the quick reply. I can't update my service file to match the one from the installation script exactly because I have different file paths that would have to be included there.
Is that also the fact that my service file also the reason Moonraker refuses to write any files?

`2023-02-02 21:52:46,260 [web.py:log_exception()] - Uncaught exception POST /server/files/upload (10.1.10.87)
HTTPServerRequest(protocol='http', host='voron01:5001', method='POST', uri='/server/files/upload', version='HTTP/1.0', remote_ip='10.1.10.87')
Traceback (most recent call last):
  File "/usr/local/klipper/.moonraker-env/lib/python3.10/site-packages/tornado/web.py", line 1681, in _execute
    result = self.prepare()
  File "/usr/local/klipper/moonraker/moonraker/app.py", line 881, in prepare
    fm.check_write_enabled()
  File "/usr/local/klipper/moonraker/moonraker/components/file_manager/file_manager.py", line 226, in check_write_enabled
    raise self.server.error(
utils.ServerError: Write access is currently disabled.  Check notifications for warnings.
2023-02-02 21:52:46,262 [app.py:log_request()] - 500 POST /server/files/upload (10.1.10.87) [_TRUSTED_USER_] 2.65ms`

Arksine, 2023-02-02:
Write access is disabled because Moonraker cannot complete the update.  If you have custom folder locations outside of the home directory you will need to manually update the installation.
- Decide where the datapath should be located and create the base folder structure.  Within the `data_path` you can create symbolic links to your existing `gcodes`, `config,` `database`, and `logs` paths. 
- Create the systemd folder in the `data_path` and `moonraker.env` within it.  The arguments within the env file should specify the location of `moonraker.py`, along with the appropriate paths for `-d`, `-c`, and `-l`.
- Update the sytemd service file using the documented service file as a template, customizing the paths to fit your installation

I recommend studying the [install docs](https://moonraker.readthedocs.io/en/latest/installation/), with a specific focus on the `Data Folder` and `Service File` sections.

If none of this is appealing  your other option would be to revert to a commit prior to when #491 was merged.

sboschert, 2023-02-02:
I did all of those steps. When starting Moonraker, it logs the correct (new) directory paths as set in the env file. Yet Moonraker still seems to think it has to update that service file. Is there a way to manually (and maybe temporarily) override this check so that I can get back to a valid installation state?

sboschert, 2023-02-02:
Update: I found out what my error was. I had a slightly different service description. Now the sudo notification disappeared and I am back to moving directories around. Anyways, thanks for pointing me in the right direction.

closed 2023-02-02 by sboschert