Arksine/moonraker #606: virtual_sdcard ignores location i set in config file.

legend069, 2023-02-03:
### What happened

i've just wiped the SD card to start again,
installed everything copied over config files ect.
setup Samba for file sharing.
sliced a file sent to the dir "/home/pi/Klipper_uploads"
checked fluidd under jobs tab and it's not there.

my Samba link was working fine before the SD card wipe. but now moonraker says a "warning" and over rides what i've set.

### Client

Fluidd

### Browser

Firefox

### How to reproduce

constant issue.

### Additional information

[moonraker(14).log](https://github.com/Arksine/moonraker/files/10579292/moonraker.14.log)
[klippy(24).log](https://github.com/Arksine/moonraker/files/10579305/klippy.24.log)
[image]
[image]
[image]
[image]

Arksine, 2023-02-03:
Moonraker requires that Klipper be configured to use the gcode path in its `data folder`.  To correct the issue you must follow the instructions in the warning and configure Klipper's `[virtual_sdcard]` section in `printer.cfg` to:
  ```
  [virtual_sdcard]
  path: /home/pi/printer_data/gcodes
  ```
If you want to share that folder in Samba then you need to configure your share to `/home/pi/printer_data/gcodes`

If you absolutely want to use `/home/pi/klipper_uploads` you can replace the `/home/pi/printer_data/gcodes` folder with a symbolic link to `/home/pi/klipper_uploads`.

legend069, 2023-02-05:
thank you, i was able to fix it with a symbolic link.

the main reason i opened this issue was because i get the "warning" of the error. but then it ignores what i've set regardless.
can the "warning" text be adjusted so it will say that it will override what the user has set in ```[virtual_sdcard]``` since it does that anyway.

Arksine, 2023-02-05:
The warning is accurate, it tells the user what the issue is and how to resolve it.  Moonraker does not override the location set in `virtual_sdcard`, however if the virtual sdcard is not configured with the correct path then Moonraker cannot manage the files in that folder.

closed 2023-02-05 by Arksine