Arksine/moonraker #609: "@" does not work in Tasmota passwords.

raldone01, 2023-02-09:
### What happened

I used the following password: Sqodc&@7DkPn98

It failed with 401.

### Client

Mainsail

### Browser

Other or N/A

### How to reproduce

Set the following password on the tasmota device: Sqodc&@7DkPn98

I used the following configuration:
```conf
[power printer_plug]
type: tasmota
address: 172.24.5.5
password: Sqodc7DkPn98
off_when_shutdown: True
on_when_job_queued: True
locked_while_printing: True
restart_klipper_when_powered: True
restart_delay: 1.
```

### Additional information

Moonraker log:

```
2023-02-09 22:57:31,276 [power.py:init_state()] - Device Init Error: printer_plug
tornado.httpclient.HTTPClientError: HTTP 401: Unauthorized

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/components/power.py", line 441, in init_state
    state = await self._send_status_request()
  File "/home/pi/moonraker/moonraker/components/power.py", line 965, in _send_status_request
    res = await self._send_tasmota_command("info")
  File "/home/pi/moonraker/moonraker/components/power.py", line 962, in _send_tasmota_command
    return await self._send_http_command(url, command)
  File "/home/pi/moonraker/moonraker/components/power.py", line 476, in _send_http_command
    response.raise_for_status(
  File "/home/pi/moonraker/moonraker/components/http_client.py", line 339, in raise_for_status
    raise ServerError(msg, code) from self._error
utils.ServerError: Error sending 'tasmota' command: info
```

Arksine, 2023-02-10:
I'll need you to attach the full moonraker.log.  FWIW, the config snippet you posted does not contain an @.

raldone01, 2023-02-10:
Yes my bad that was the password that worked. I can provide you a full log later.

raldone01, 2023-02-11:
I had time to do further testing. Corrected the issue.
I think the & is causing the issue.
Passwords without & work just fine with moonraker.

Arksine, 2023-02-11:
This should be resolved with commit 55ebe120c53c293384e69c31d5626baa9c87d2f9.

Indeed the issue was the `&`.

closed 2023-02-21 by Arksine