Arksine/moonraker #611: after loading history - there are already 2 selected jobs

fused3dprints, 2023-02-13:
### What happened

[image]

I just noticed that i automatically have 2 selected jobs whenever i load history tab but the details are now shown.
I can export the jobs (via CSV) and can confirm these are old jobs i printed.
Tried to query via history api and they are not showing. 

### Client

Mainsail

### Browser

Chrome, Microsoft Edge

### How to reproduce

just need to open the history tab

### Additional information

[print_history.csv](https://github.com/Arksine/moonraker/files/10721775/print_history.csv)
[API_out.txt](https://github.com/Arksine/moonraker/files/10721794/API_out.txt)

attaching the CSV file and API history output for reference.

fused3dprints, 2023-02-13:
Thanks for the quick response. 
I'm really new to this and will try to open a BUG report in the mainsail github.

Arksine, 2023-06-11:
I am going to close this as it seems like the discussion has concluded.

closed 2023-06-11 by Arksine