Arksine/moonraker #612: Ability to suppress moonraker warnings by ID

trevjonez, 2023-02-13:
### Is your feature request related to a problem? Please describe

Because I have multiple printers that all share the bulk of their configuration I store all that in a single git repo that is then cloned to the PI.

From there I symlink from `~/klipper_config/printer.cfg` to `~/klipper-configs/fysetc_s6_bear_2.cfg`.
This has worked great for sharing updates to my configuration across machines quickly and continues to work well today. 

However, it causes a large undismissable banner to take up a large chunk of valuable real-estate at the top of all of my fluidd GUIs. 

[image]

It is unclear if this will also impact mainsail, though I should know in the near future once my voron build is ready to boot up.

### Describe the solution you'd like

It would be nice if we could configure moonraker to ignore warnings by ID.

roughly:
```
[server]
suppress_warning:
  klipper_config
```

Then should the need arise the user can simply comment that portion of the config while diagnosing potential issues. IE when dealing with updates that include breaking changes to the configuration file.

Arksine, 2023-02-13:
This is a bug IMO, Moonraker shouldn't warn if the config file is directly linked in the config path.  Commit a40dae2bc8fe4066da9fa193b566e77192068ab4 should correct this issue.

trevjonez, 2023-02-13:
> This is a bug IMO, Moonraker shouldn't warn if the config file is directly linked in the config path. Commit [a40dae2](https://github.com/Arksine/moonraker/commit/a40dae2bc8fe4066da9fa193b566e77192068ab4) should correct this issue.

Fix confirmed. Giant thanks on the quick turn around.

[image]

closed 2023-02-13 by trevjonez

alxpettit, 2023-12-12:
I would have benefited from the ability to suppress warnings by ID, by the way. NixOS-managed config files have to do some hacks to get Moonraker to cooperate with what are fundamentally immutable configurations, and Moonraker has recently started nagging me on my entire print farm over what seems to be an entirely reasonable setup.

Arksine, 2023-12-12:
I don't think its good idea to be able to suppress warnings it encourages users to ignore them rather than address the underlying issue.  If Moonraker is warning when it shouldn't I'm willing to address that, however I need a log to determine which warning is causing the issue.

In addition I recommend Debian or one of its derivatives.  You are certainly free to use whichever distro you like, however distros that deviate from the FHS may encounter issues or require "hacks" as you put it.