Arksine/moonraker #613: Missing or invalid secrets file prevents moonraker service from loading

zanhecht, 2023-02-14:
### What happened

When the moonraker.secrets file is missing, or a requested password is missing, instead of raising a config.error message, jinja2 throws an UndefinedError and the entire moonraker service will fail to load.

### Client

Other

### Browser

Other or N/A

### How to reproduce

1. Add a call to the secrets file (e.g. [machine] sudo_password = {secrets.sudo.password}) to moonraker.conf
2. Add a blank [secrets] section to moonraker.conf
3. Do not create a moonraker.secrets file
4. Attempt to start the moonraker service

### Additional information

```
--------------------Moonraker Log Start--------------------
data_path: /home/pi/printer_data
is_default_data_path: False
config_file: /home/pi/printer_data/config/moonraker.conf
startup_warnings: []
verbose: False
debug: False
asyncio_debug: False
is_backup_config: False
log_file: /home/pi/printer_data/logs/moonraker.log
software_version: v0.7.1-874-gbec9a66
python_version: 3.9.2 (default, Feb 28 2021, 17:03:44)  [GCC 10.2.1 20210110]
2023-02-14 10:06:32,833 [loghelper.py:__init__()] - data_path: /home/pi/printer_data
2023-02-14 10:06:32,834 [loghelper.py:__init__()] - is_default_data_path: False
2023-02-14 10:06:32,834 [loghelper.py:__init__()] - config_file: /home/pi/printer_data/config/moonraker.conf
2023-02-14 10:06:32,834 [loghelper.py:__init__()] - startup_warnings: []
2023-02-14 10:06:32,834 [loghelper.py:__init__()] - verbose: False
2023-02-14 10:06:32,834 [loghelper.py:__init__()] - debug: False
2023-02-14 10:06:32,834 [loghelper.py:__init__()] - asyncio_debug: False
2023-02-14 10:06:32,834 [loghelper.py:__init__()] - is_backup_config: False
2023-02-14 10:06:32,834 [loghelper.py:__init__()] - log_file: /home/pi/printer_data/logs/moonraker.log
2023-02-14 10:06:32,835 [loghelper.py:__init__()] - software_version: v0.7.1-874-gbec9a66
2023-02-14 10:06:32,835 [loghelper.py:__init__()] - python_version: 3.9.2 (default, Feb 28 2021, 17:03:44)  [GCC 10.2.1 20210110]
2023-02-14 10:06:32,849 [confighelper.py:read_file()] - Configuration File '/home/pi/printer_data/config/moonraker.conf' parsed, total size: 5229 B
2023-02-14 10:06:32,850 [moonraker.py:add_log_rollover_item()] - 
#################### Moonraker Configuration ####################

[server]
host = 0.0.0.0
port = 7125
max_upload_size = 1024
klippy_uds_address = ~/printer_data/comms/klippy.sock

[file_manager]
enable_object_processing = False

[authorization]
cors_domains = 
	https://my.mainsail.xyz
	http://my.mainsail.xyz
	http://*.local
	http://*.lan
trusted_clients = 
	10.0.0.0/8
	127.0.0.0/8
	169.254.0.0/16
	172.16.0.0/12
	192.168.0.0/16
[... 149 lines cut ...]
  File "/home/pi/moonraker/moonraker/components/machine.py", line 102, in __init__
    self._sudo_password = sudo_template.render()
  File "/home/pi/moonraker/moonraker/components/template.py", line 92, in render
    return self.template.render(context).strip()
  File "/home/pi/moonraker-env/lib/python3.9/site-packages/jinja2/environment.py", line 1301, in render
    self.environment.handle_exception()
  File "/home/pi/moonraker-env/lib/python3.9/site-packages/jinja2/environment.py", line 936, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "<template>", line 1, in top-level template code
  File "/home/pi/moonraker-env/lib/python3.9/site-packages/jinja2/environment.py", line 485, in getattr
    return getattr(obj, attribute)
jinja2.exceptions.UndefinedError: 'components.secrets.Secrets object' has no attribute 'sudo'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/moonraker.py", line 501, in main
    server.load_components()
  File "/home/pi/moonraker/moonraker/moonraker.py", line 231, in load_components
    self.load_component(config, component)
  File "/home/pi/moonraker/moonraker/moonraker.py", line 262, in load_component
    raise ServerError(msg)
utils.ServerError: Unable to load component: (machine)
2023-02-14 10:06:33,437 [moonraker.py:main()] - Server Shutdown
```

Arksine, 2023-02-14:
Thanks, this should be resolved in 87dba2f2e2c78267b2ce9d4a4f54d4aa0bc7ed8a.

closed 2023-02-21 by Arksine

JaysenSingh, 2023-12-27:
Hi all, can anyone help with this error please. I've updated moonraker via fluidd notification and then this error happened. [image]

Arksine, 2023-12-28:
You will need to provide `moonraker.log`.  The error suggests that you have a version of pylmdb installed that is very old, older than the version initially added with Moonraker.

JaysenSingh, 2023-12-28:
Hi Eric,
Thank you for your reply, I have attached my log file for viewing.
Any help is greatly appreciated.
Thanks,Jay

Arksine, 2023-12-28:
@JaysenSingh I don't believe that the log attached correctly.  That said, others are experiencing this issue, please follow #781.