Arksine/moonraker #614: Fluidd won't load thumbnails without restarting moonraker service

trevjonez, 2023-02-14:
### What happened

When I add a new gcode file to my directory tree the details and thumbnail will not load in fluidd until I restart the moonraker service.

### Client

Fluidd

### Browser

Chrome

### How to reproduce

So before I add a new gcode file looks like this:
[image]

Then when I add a new gcode file to the directory and hit refresh on the jobs panel:
[image]

Then I restart moonraker and push refresh on the jobs panel again:
[image]

### Additional information

While writing this up I realized that maybe the issue is that I don't upload files via the moonraker API but actually write them to my NAS server, then each of the klipper hosts have and folder mount via autofs. 

Then the directory is symbolic linked: `/home/pi/gcode_files -> /home/pi/nas/gcodes`

Looks like if I configure the slicer to upload directly the meta data does load.
[image]

BUT it only loads on the machine that it was uploaded through.

[image]

Is there a way to tell moonraker to load missing file metadata when running a refresh from the fluidd panel?

Arksine, 2023-02-14:
Please attach `moonraker.log`.  I suspect the filesystem you have your gcode files on does not support inotify, which is a requirement for Moonraker to reliably manage files.  Inotify allows Moonraker to watch certain paths and react to changes, even if they are sourced outside of Moonraker.  Frankly I'm surprised your attempt to upload via the slicer worked at all, as Moonraker synchronizes inotify events with API requests.  If the FS doesn't support inotify the request should timeout with an error.

There is no API for scanning or requesting metadata.

trevjonez, 2023-02-14:
Ok so you are spot on about inotify.

Moonraker log has no errors for the file system in it which I think matches what I am seeing via `inotify-tools` from the CLI. Pretends to work, but nothing is being sent. Effectively the file system is lying to moonraker. Here is what the CLI looks like after an edit from my desktop as well as the same PI.

[image]

Doing the same command from my Ubuntu box it seems to work fine. So I suspect this is just an issue with how my autofs mounts are configured on the PI and I'll need to compare the configurations.
[image]

All that said, this is clearly not an issue with moonraker. Thank you for the prompt response!

trevjonez, 2023-02-17:
> Doing the same command from my Ubuntu box it seems to work fine.

So after more digging it looks like a cifs mount just can't correctly send inotify events. The reason (or so I assume) it seemed to work on my ubuntu box is that the inotifywait and edits to files were on the same box.

Would you be willing to consider this issue a feature request to add an RPC to emulate whatever the inotify is triggering so that I can write a cron job to workaround my issue?

Arksine, 2023-02-18:
In the short term you may be able to simulate an inotify "modify" event by appending whitespace to the file from the Pi:

```
echo -e "\n" >> ~/printer_data/gcodes/my_file.gcode
```

The file being touched must be in a directory that was created before Moonraker starts.  Moonraker won't receive directory events, and thus cannot add watches for new nodes.

I am willing to add an endpoint that scans metadata upon request.    I think the best long term solution for CIFS mounts would be to implement a simple "polling" observer may be configured to use in place of inotify.

Arksine, 2023-02-18:
FYI, in a brief test the above worked on my system.  The inotify modify event will get generated if the same machine hosting Moonraker modifies the file on the cifs mount.  Presumably you could implement your own rudimentary polling observer then modify the file.  If the mount is shared by multiple Pis you may need to write a lockfile first since the writes aren't atomic.  If you want to get fancy I'm sure you can write a byte then remove it before closing to trigger the event.

My understanding is that this isn't a limitation of the server.  The linux cifs client doesn't support inotify events, been like that for almost 2 decades.

trevjonez, 2023-02-19:
> My understanding is that this isn't a limitation of the server. The linux cifs client doesn't support inotify events, been like that for almost 2 decades.

That is about as far down the rabbit hole I made it before coming to reopen. 

The echo workaround is prefect for now. Once I find some spare time I am going to play with using fuse instead. Seems there may be a path to get the filesystem to just behave well enough that none of this is needed.
https://github.com/libfuse/libfuse/wiki/Fsnotify-and-FUSE

siparker, 2023-03-03:
@trevjonez @Arksine 

There is no need for the endpoint to be added. it already exists and is documented.

posting to http://xxx.xxx.xxx.xxx/server/files/metascan

{
    "jsonrpc": "2.0",
    "method": "server.files.metascan",
    "filename": "share/test/1.gcode",
    "id": 3545
}

in fact as i screwed it up slightly and put a full path filename in. it ran for ages. timed out twice. and when i now check the share folder all the files in there have meta data showing...

Not sure why that happened. but its very useful. i will see if i can reproduce that.

So just need to find the best way to monitor the folder for changes, send a request to that endpoint and we can share files from wherever we like regardless of inotify compatibility

closed 2023-12-12 by trevjonez