Arksine/moonraker #626: passing event_args in path or url appears to be broken

aSocialMenace, 2023-03-04:
### What happened

I am trying to set up a moonraker notifier to send me a telegram when an event happens. My pi does not have a web cam attached as i power my pi through the TFT header on my skr mini e3 v3 and im concerned the webcam would draw to much power. anyways i would like to configure the notifier to send a gcode thumbnail instead of a snap shot. Here is what i have

[notifier print_start]
url: tgram://{bottoken}/{ChatID} (properly configured of course)
events: started
body: Your printer started printing '{event_args[1].filename}'
attach: ~/printer_data/gcodes/.thumbs/{event_args[1].filename}.png

I have tested the attach section with the path of a known gcode file like

attach: ~/printer_data/gcodes/.thumbs/voroncube.png

and it will send the voroncube png however when i swap the voroncube.png out with the {event_args[1].filename}.png all i get is a telegram with the body and no image. 

i have also tried using a url to get this png with the same results 

attach: http://ogender3pro.local/server/files/gcodes/gcodes/.thumbs/{event_args[1].filename}.png

### Client

Mainsail

### Browser

Other or N/A

### How to reproduce

insert the event_args in the path or url in the attach field of the notifier section of the moonraker.config file

attach: ~/printer_data/gcodes/.thumbs/{event_args[1].filename}.png

hmortensen83, 2023-03-04:
If filename includes extension like **voroncube.gcode** you will end up with **voroncube.gcode.png**

aSocialMenace, 2023-03-04:
but when the body calls the {event_args[1].filename} variable it doesnt attach the .gcode.
this line
 
body: Your printer started printing '{event_args[1].filename}

give me this text in my message

Your printer started printing 'name of file printing'

so that variable can't be adding a file extension..  right? I mean i could be missing something..

aSocialMenace, 2023-03-04:
ok had a chance to take a stab at fixing this error. i added [:-6] to the end of the variable to subtract the 6 characters of the .gcode and just to double check that it would work i added it to the body as well. my new code looks like this

[notifier print_start]
url: tgram://{bottoken}/{ChatID} (properly configured of course)
events: started
body: Your printer started printing {event_args[1].filename[:-6]}.png
attach: ~/printer_data/gcodes/.thumbs/{event_args[1].filename[:-6]}.png

When I print voroncube.gcode i get a telegram message that says 

Your printer started printing voroncube.png

but still no image. now if i were to replace  {event_args[1].filename[:-6]} with the file name voroncube.png like this

attach: ~/printer_data/gcodes/.thumbs/voroncube.png

the message sends the voroncube.png with the body message... but obviously that means it will send that same image no matter what im printing... but my path is correct and my variable is correct.. the concatenation is working as expected.. 

i'm stumped here.. any ideas?

Arksine, 2023-03-04:
I took a look at this, the issue is that the `attach` option does not accept templates.  Thus, the value for `attach` ends up not existing.

I'll look into adding support for templates for the attachment when I have the opportunity.

aSocialMenace, 2023-03-04:
Cool! Thanks you :)

Arksine, 2023-03-05:
Commit f29c896c4a6a54917711bf23eaec813961bd6821 should allow templates to resolve the correct file in the `attach` option.

Additionally I noticed that Apprise supports multiple attachments.  I added the ability to specify multiple items in the `attach` option.  When doing so, each item should be separated by a newline.

aSocialMenace, 2023-03-05:
Tested it!! works like a charm!!! 👍 👍 👍  Thank you!!!

closed 2023-03-05 by Arksine