Arksine/moonraker #632: Port numbers same one 2nd and 3rd printers.

mikeoverbay, 2023-03-19:
### What happened

Maybe this has been reported.
The port numbers are the same for my 2nd and 3rd printers.
Both where set at 2526. Only the first printer was set and 7125.

```
[server]
host: 0.0.0.0
port: 7126 <--- being duplicated on 2nd and 3rd
klippy_uds_address: /home/sumthin/Pro_data/comms/klippy.sock
```

### Client

Mainsail

### Browser

Chrome

### How to reproduce

Build 3 klipper installs
But 3 Moonraker installs.
install mainsail.
check moonraker.conf for each klipper install.

Arksine, 2023-03-19:
It isn't clear what you are reporting.  It is up to the user to configure the correct port after installation.

mikeoverbay, 2023-03-22:
the config files for printers 2 and 3 have the same port number. Maybe this has something to so with the installer? https://github.com/th33xitus/kiauh

Arksine, 2023-03-29:
Sorry for the delay. Yes, this appears to be an issue with kiauh, as Moonraker's basic installer provides no method for setting the port, it is expected that the user will customize their configuration after installing if necessary.

closed 2023-03-29 by Arksine