Arksine/moonraker #637: Moonraker power control don't work properly on Raspberry pi Zero 2

tiomel, 2023-03-23:
### What happened

I don't know if I'm raising this question in the right place, I apologize if I'm not. The fact is that I have tried to use a double relay to turn my printer on and off (voron2.4) that works with a PiZero2 with the fully updated official image.

I have added the configuration so that moonraker uses the gpio22, and when configuring it and restarting the whole relay it stays permanently activated. Pressing the button from the interface has no effect on the GPIO, the relay remains activated. I also configured another pin in the klipper configuration to control the lights, same result, the moment I defined it in my printer.cfg the relay is permanently activated.

I have tried numerous solutions but none of them have helped me, I don't know if it is a problem with moonraker, with permissions, or with p zero itself.

To expand the information my installation was made from the official image, everything else installed with kiahu, everything is up to date, I configured my host as MCU and defined it in my configuration.

The output of GPIO info attached on image.

### Client

Mainsail

### Browser

Chrome

### How to reproduce

all explained above

### Additional information

[image]
[moonraker_1.log](https://github.com/Arksine/moonraker/files/11055110/moonraker_1.log)
[moonraker_config.txt](https://github.com/Arksine/moonraker/files/11055117/moonraker_config.txt)

wattsie, 2023-03-24:
My $0.10 worth..

First, make absolutely sure nothing else is trying to control that pin. 
Do NOT install WiringPi. 
Do not install anything but Klipper and Moonraker.
Remove all config, and restarting it so it releases the gpio pin.

Then try manually toggling the pin.
(commands below assume default install of MainsailOS, other OS be different)

To see all pins 
`raspi-gpio get` 

To set pin to output mode and in off state
`raspi-gpio set 22 op,dl`
Relay should be off.

Then set to on state
`raspi-gpio set 22 dh`
Relay should be on.

Toggle the "dl" and "dh" to drive pin high and low respectively.

This will simply confirm you are able to toggle the relay and that the pin matches the pin the relay is connected to. 

Once you know you can control the pin, and it works as expected...

Add back in the power pin config...

Then enable debug logging in by adding the following option to the [server] section in moonraker.conf

```
[server]
enable_debug_logging: True
```
then restart the moonraker service
```
sudo service moonraker restart
```

If it doesnt work, then post your log files again. Just be sure you can manually drive the relay from the command line.

Again check nothing else is locking access to the GPIO, like wiringpi etc.
WiringPi uses different pin numbering scheme than libgpiod. 
Make sure your relay is actually connected to right pin as noted on [this pinout](https://pinout.xyz/).

Hope this helps.

tiomel, 2023-03-24:
Well this is a bit embarrassing and probably due to my lack of knowledge, the fact is that this morning I found help to use those commands that you show me above, the one that shows the status of a certain GPIO. I noticed that the state changed when I activated the button in Mainsail, so I had confirmation that it was really working. So I changed my relay.

The one that I have been using and that did not work for me: https://www.emartee.com/Images/websites/emartee.com/42030.jpg

And this is the one I'm using now and it works correctly:
https://media-cdn.elecrow.com/media/webp_image/catalog/product/cache/f8158826193ba5faa8b862a9bd1eb9e9/1/4/14187984670_1.webp

I don't know the reason, if someone could clarify it, we may help other people with this problem.

wattsie, 2023-03-25:
I would hazard a guess that being 5V based, the SSR based one is a trigger on low, so floating at 3.3v would be OFF, and pulled low to ground would trigger ON.
Whereas the Wrobot relay version may require 5V on the trigger on or off, and the max 3.3v of the RPI is not sufficient to trigger.

But again, this is a guess based on quick look at the device.

Glad you got one of the units working for your needs.

If you are happy, please feel free to close this issue.

Cheers.

tiomel, 2023-03-25:
Thank you for taking the time to write, what was happening to the YwRobot relays is that they were permanently activated, now it is working with these new relays so I can close this, although I still have to study these relays a bit. Thanks my friend.

closed 2023-03-25 by tiomel