Arksine/moonraker #648: MQTT Connection ERROR

smilecaozheng, 2023-04-29:
### What happened

Due to my MQTT server was started by Docker, Whenever I start Moonraker, the MQTT server has not yet started. So It reported an error. Can I set the [mqtt] option in moonrake.config ,To wait for the MQTT server to start successfully? If not, can we join in the future? 

Ps: I noticed the following code in the Python code, and I placed the screenshot at the end.

Error log:
[image]

### Client

Fluidd

### Browser

Microsoft Edge

### How to reproduce

Startup sequence：
moonraker[mqtt client] -> docker[mqtt server] -> MQTT Connection Error

In this sequence, it can be seen from the server that another device is connected, but Moonraker is not connected.
[image]

Restart moonrake：

[image]

Arksine, 2023-05-07:
Moonraker should attempt to reconnect if the initial connection fails.  The full `moonraker.log` needs to be attached for me to diagnose further.

smilecaozheng, 2023-05-10:
[moonraker.log](https://github.com/Arksine/moonraker/files/11440965/moonraker.log)
Attach log file

Arksine, 2023-05-10:
A few things I can determine from the log:
1) It appears that Moonraker's source has been modified.  It's difficult for me to provide reliable feedback on modified code as the modifications themselves can be the source of the issue.
2) The traceback reported in the OP isn't reproduced in the log attached.
3) It does appear that on some attempts that the client is connecting, however the broker isn't responding.  This issue would be outside of Moonraker's scope.

smilecaozheng, 2023-05-11:
Here's an explanation：
1.Indeed, this has added some of my own components. I will remove this and resend the log.
The current state should be uncontaminated.
[image]

2.Will be retested in sequence.
3.This part doesn't seem to be understood, it's like this: when Moonraker starts, the broker doesn't start, it only starts about 2 minutes after Moonraker starts. Does this mean that if Moonraker starts without discovering the broker, it will throw a timeout exception and will not attempt to connect again.

(1)
moonraker[mqtt client] ->At this time, an error should have been reported

[moonraker (1).log](https://github.com/Arksine/moonraker/files/11452454/moonraker.1.log)

docker[mqtt server] -> another device is connected, but Moonraker is not connected.
[image]
Please ignore Chinese

(2)
moonraker[mqtt client] restart
[moonraker (2).log](https://github.com/Arksine/moonraker/files/11452479/moonraker.2.log)

[image]
MzEwNjA4MTU0NDEzMDU1N ：This device is a klipper device

smilecaozheng, 2023-07-27:
Solution: Install the mqtt server directly without using Docker

closed 2023-07-27 by smilecaozheng