Arksine/moonraker #653: Component 'dbus_manager' failed to load with error:

acidcoolxxx, 2023-05-09:
### What happened

Hi

can't find any helpful things on this. 

2023-05-09 22:38:14,835 [server.py:_initialize_component()] - Component [dbus_manager] failed post init
Traceback (most recent call last):
  File "/opt/moonraker/moonraker/server.py", line 222, in _initialize_component
    await ret
  File "/opt/moonraker/moonraker/components/dbus_manager.py", line 65, in component_init
    self.polkit = await self.get_interface(
                  ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/moonraker/moonraker/components/dbus_manager.py", line 109, in get_interface
    ret = await self.get_interfaces(bus_name, bus_path,
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/moonraker/moonraker/components/dbus_manager.py", line 121, in get_interfaces
    introspection = await self.bus.introspect(bus_name, bus_path)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/venv/lib/python3.11/site-packages/dbus_next/aio/message_bus.py", line 216, in introspect
    return await asyncio.wait_for(future, timeout=timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/tasks.py", line 479, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
  File "/opt/venv/lib/python3.11/site-packages/dbus_next/message_bus.py", line 195, in reply_notify
    BaseMessageBus._check_method_return(reply, err, 's')
  File "/opt/venv/lib/python3.11/site-packages/dbus_next/message_bus.py", line 611, in _check_method_return
    raise err
  File "/opt/venv/lib/python3.11/site-packages/dbus_next/aio/message_bus.py", line 365, in _message_reader
    if self._unmarshaller.unmarshall():
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/venv/lib/python3.11/site-packages/dbus_next/_private/unmarshaller.py", line 304, in unmarshall
    self._unmarshall()
  File "/opt/venv/lib/python3.11/site-packages/dbus_next/_private/unmarshaller.py", line 243, in _unmarshall
    self.read(16, prefetch=True)
  File "/opt/venv/lib/python3.11/site-packages/dbus_next/_private/unmarshaller.py", line 91, in read
    raise EOFError()
EOFError
2023-05-09 22:38:14,835 [server.py:add_warning()] - Component 'dbus_manager' failed to load with error: 

### Client

Mainsail

### Browser

Chrome

### How to reproduce

dont know

Arksine, 2023-06-11:
This looks like an issue on your system.  Moonraker does not yet officially support Python 3.11, however I haven't heard of any specific compatibility issues.   Without the full `moonraker.log` attached I can't provide much more feedback.

mkuf, 2023-07-01:
>  Moonraker does not yet officially support Python 3.11, however I haven't heard of any specific compatibility issues. 

The `pyproject.toml` indicates a compatibility w/ 3.11.  
Could you specify which Version should be used for best compatibility?

Arksine, 2023-07-03:
The versions in `pyproject.toml` are stubbed, I do plan to test on 3.11 before releasing a packaged version of Moonraker.   Moonraker itself shouldn't have a problem with 3.11, however some of the dependencies, particularly `dbus_next`, may have an issue.

If that is confusing I can remove it for now.  Moonraker should run fine on python 3.10.

That said, I really need a full log to further analyze the issue.  With the specific case of running Moonraker inside a container it is likely that you want to disable features that rely on dbus.  Specifically, the `provider` option in `[machine]` needs to be properly configured, it defaults to `systemd_dbus` but I doubt that `systemd` is available.  You probably don't want to configure the `update_manager` in a container, but if you do `enable_system_updates` should be set to `False`.

mkuf, 2023-07-05:
Thanks, for docker use, systemd is available inside the container and interfaces with the hosts systemd process.  
For this to work, the provider is set to `systemd_cli`.

I couldn't reproduce the issue with either python 3.11 or 3.10, so maybe @stockholmux can share a full log file for further troubleshooting.

stockholmux, 2023-07-05:
Yeah, here is the log output. 

One note, I did set `enable_system_updates: False` in the `[update_manager]` section of the `moonraker.conf` in efforts to fix the issue, and you can see some complaining in the log file about that as well (it didn't help).

[moonraker.log](https://github.com/Arksine/moonraker/files/11962643/moonraker.log)

Arksine, 2023-07-22:
FWIW, commit 8eddec88cbcdf0515e086367630068df8379375c should resolve this issue.  You will definitely want to remove the `update_manger` from your config, in most containers it is likely to result in nothing but errors.

Arksine, 2023-10-27:
I'm going to close this as resolved.  Thanks.

closed 2023-10-27 by Arksine