Arksine/moonraker #657: Network connection problems after moonraker update

ptrstourac, 2023-05-23:
### What happened

Today i updated moonraker to the latest version. Since that I have problem with network connection timing out randomly for few minutes and then going back on. My router shows Intel NUC running the klipper as connected for the whole time, but I'm not able to even ping the device (nor open Mainsail in the browser or ssh to it), every time returns timeout. 

Interval in which this happens is random, connection drops for 5 or so minutes and then goes back on for some time. Doesn't matter if the printer is idle or printing.

Attaching moonraker.log and corresponding part of log from NetworkManager.

### Client

Mainsail

### Browser

Firefox

### How to reproduce

I'm using Intel NUC 7PJYH2 running Ubuntu 22.04.2 LTS. Using two instances of Klipper and Moonraker for my 2 printers. NUC is connected via WiFi. Device has reserved IP and means for connection from outside provides ZeroTier VPN. 

### Additional information

[moonraker.log](https://github.com/Arksine/moonraker/files/11548343/moonraker.log)
[NetworkManager Journal.txt](https://github.com/Arksine/moonraker/files/11548351/NetworkManager.Journal.txt)

Arksine, 2023-05-24:
Moonraker doesn't manage or in any way control the underlying network hardware, an update to Moonraker should not impact the connection.  It sounds like you are experiencing a wifi issue, if you performed a system update its possible that a kernel update is the cause.

If possible I would recommend trying an alternative network connection such as a wired connection or a WIFI USB dongle.  If this resolves the issue it would confirm that it is an issue with the onboard wifi.

ptrstourac, 2023-05-30:
Connection issues resolved, not related to Moonraker.

closed 2023-05-30 by ptrstourac