Arksine/moonraker #662: Klipper configuration already exists, Update manager

AmyTheCute, 2023-05-29:
### What happened

Hi, I'm trying to make our 3DPrinters update from our repos (slightly modified firmware) in the update manager screen.
I followed the instruction, configured and put our own repos in the config file, put moonraker in debugging mode and yet, we get errors about the config file

This is the config file:
```
[server]
host = 0.0.0.0
port = 7125
klippy_uds_address = /home/biqu/printer_data/comms/klippy.sock

[authorization]
trusted_clients = 
	10.0.0.0/8
	127.0.0.0/8
	169.254.0.0/16
	172.16.0.0/12
	192.168.0.0/16
	FE80::/10
	::1/128
cors_domains = 
	http://*.lan
	http://*.local
	https://my.mainsail.xyz
	http://my.mainsail.xyz
	https://app.fluidd.xyz
	http://app.fluidd.xyz

[octoprint_compat]

[history]

[update_manager]
refresh_interval = 168
enable_auto_refresh = True

[update_manager klipper]
type = git_repo
channel = dev
origin = https://github.com/Desuuuu/DGUS-reloaded-Klipper (example repo. this comment isn't in the config file)
path = ~/klipper
primary_branch = HS3-Production-Utilities
is_system_service = True
managed_services = klipper
```

### Client

Other

### Browser

Other or N/A

### How to reproduce

Add klipper to the moonraker.conf
put moonraker in debug mode as stated by manual
restart system/moonraker

### Additional information

[moonraker-2.log](https://github.com/Arksine/moonraker/files/11587088/moonraker-2.log)

Arksine, 2023-06-11:
When the debug flag is added validation checks in the `update_manager` are disabled, allowing updates on unofficial remotes.  There is no need to add an override to `moonraker.conf`.

AmyTheCute, 2023-06-11:
> When the debug flag is added validation checks in the `update_manager` are disabled, allowing updates on unofficial remotes. There is no need to add an override to `moonraker.conf`.

But how do i let moonraker know what repo I intend to use?

Arksine, 2023-06-11:
Moonraker will update the branch that is currently checked out.  With the debug flag enabled it won't reject an unofficial remote url.   Clone your fork and configure Moonraker to connect to it.  After the initial connection Moonraker will detect its install location and it should be available to update.

Alternatively you can add your fork's remote to an existing installation and checkout a branch from it.  With this method you may need to update the `git config` to make sure that the branch you want to use points to the desired remote.

FWIW, I intend to remove the need for the `debug` flag to use this functionality in the near future.  I will provide an update when I get a chance to finish and test the changes.

Arksine, 2023-07-09:
FYI, it is now possible to update forks without enabling the debug flag.  No additional configuration is required.

closed 2023-07-09 by Arksine