Arksine/moonraker #665: Print history file exists test has false positives

Riksu9000, 2023-05-31:
### What happened

I tend to upload gcode files with similar names to my printer, replacing the old files. The issue is that when a file is replaced, the print history in Mainsail still shows that a file exists for a print, even though the file is replaced and is a different print.

I posted this issue first in Mainsail and was redirected here. You'll find more information there.

https://github.com/mainsail-crew/mainsail/issues/1402

### Client

Mainsail

### Browser

Firefox

### How to reproduce

Print two different gcode files that have the same filename

Arksine, 2023-06-01:
Thanks.  This should be resolved in commit d6231634db27a86c3674c051b991b1276c0d1397.

Riksu9000, 2023-06-01:
The fix seems to work well. Thanks!