Arksine/moonraker #668: JSON-RPC Request Error: -32601 Method not found

FURONGsun, 2023-06-06:
### What happened

I install the klipper moonraker and fluidd by Klipper install helper in "https://github.com/th33xitus/kiauh" at my TV BOX which ARMBIAN running on it.When I install it currecttly on it,It can work at first.But when I restart the TV BOX,It cound not working,but the info on the screen of my printer tell me that the klipper on the BOX was working.but the fluidd tell me Klippy: Disconnected.When I  check the log of klippy, it didn't give me any warming and issue.It end at"Starting heater checks for extruder" when I check the log of moonraker,it tell me "2023-06-06 14:06:28,528 [common.py:build_error()] - JSON-RPC Request Error: -32601Method not found" So I guess I need to find the answer where I am.Cound anyone give me a answer.I will be deeply grateful.

### Client

Fluidd

### Browser

Chrome

### How to reproduce

I installl it,and it can work.
but if I reboot the system,i can not work until I remove all of the software and install it again.

FURONGsun, 2023-06-06:
It seens that my klipper is not working

closed 2023-06-06 by FURONGsun

thijstriemstra, 2023-06-23:
I'm having the same issue:
```
2023-06-23 22:29:24,792 [common.py:build_error()] - JSON-RPC Request Error: -32601
Method not found
```

You closed the ticket @FURONGsun but could you let us know what solved the problem?

Piezoid, 2023-06-23:
> You closed the ticket @FURONGsun but could you let us know what solved the problem?

This error is likely moonraker failing to connect to klippy's unix socket. This can be confirmed by the absence of `[server.py:load_component()] - Component (klippy_apis) loaded` in moonraker.log

You'll have to:
- ensure that klippy is running: `systemctl status klipper.service`,
- check that the moonraker config parameter `[server] klippy_uds_address` is a valid path to klippy's socket (the one after the `-a` flag on klippy's command line, as found in the `/etc/systemd/system/klipper.service` file).

These indications may differ depending on the installation method used.

thijstriemstra, 2023-06-23:
> This can be confirmed by the absence of [server.py:load_component()] - Component (klippy_apis) loaded in moonraker.log

```
$ grep -Rn "Component (klippy_apis) loaded" printer_data/logs/moonraker.log
446:2023-06-24 00:02:35,211 [server.py:load_component()] - Component (klippy_apis) loaded
```

> check that the moonraker config parameter [server] klippy_uds_address is a valid path to klippy's socket (the one after the -a flag on klippy's command line, as found in the /etc/systemd/system/klipper.service file).

thanks for pointing me in the right direction @Piezoid! It was indeed missing the `-a` parameter, not being passed to `klippy.py`. I ended up adding this to the klipper systemd service:

```console
-a /home/foo/printer_data/comms/klippy.sock
```

And adding that path to the moonraker config:

```ini
# The address of Unix Domain Socket used to communicate with Klippy. This
# option accepts Jinja2 Templates, where the configured data path is
# passed to the template context, for example:
#   klippy_uds_address: {data_path}/comms/klippy.sock
# Default is /tmp/klippy_uds.
klippy_uds_address: {data_path}/comms/klippy.sock
```

so basically, if you run into the dreaded `[common.py:build_error()] - JSON-RPC Request Error: -32601 Method not found` error, make sure the klippy unix socket is created.

psi-4ward, 2023-06-29:
I run into that issue as well cause of wrong configuration of `klippy_uds_address`.

1. Look if `klipper` runs:  `sudo systemctl status --no-pager --full klipper`
```
● klipper.service - Klipper 3D Printer Firmware SV1
     Loaded: loaded (/etc/systemd/system/klipper.service; enabled; vendor preset: enabled)
     Active: active (running) since Thu 2023-06-29 10:59:02 BST; 6min ago
       Docs: https://www.klipper3d.org/
   Main PID: 5642 (python)
      Tasks: 6 (limit: 1857)
        CPU: 7.534s
     CGroup: /system.slice/klipper.service
             └─5642 /home/pi/klippy-env/bin/python /home/pi/klipper/klippy/klippy.py /home/pi/printer_data/config/printer.cfg -I /home/pi/printer_data/comms/klippy.serial -l /home/pi/printer_data/logs/klippy.log -a /home/pi/printer_data/comms/klippy.sock

Jun 29 10:59:02 klipper systemd[1]: Started Klipper 3D Printer Firmware SV1.
```
Notice the `active (running)` state and the `-a /home/pi/printer_data/comms/klippy.sock` parameter

2. Check if the socket exists
```shell
ls -l /home/pi/printer_data/comms/klippy.sock
srwxr-xr-x 1 pi pi 0 Jun 29 10:59 /home/pi/printer_data/comms/klippy.sock
```

3. Check if the value of `klippy_uds_address` matches the socket path.

4. Restart moonraker and start printing ;)

Riik, 2024-05-16:
Thanks. I had this same issue. Posting here just in case it'd help somebody. It was working fine before, but for some reason in `~/printer_data/config/moonraker.conf` the line 
```
klippy_uds_address: /home/<my_homedir_name>/printer_data/comms/klippy.sock
```
had changed into
```
klippy_uds_address: /home/pi/printer_data/comms/klippy.sock
```

Easy fix with help of this thread, but no idea why this seemingly out of the blue happened. Only thing I can think of was my Pi was unresponsive and I power cycled it. Maybe the file was replaced with a default one?