Arksine/moonraker #671: Webcam config not working as supposed

lettore, 2023-06-10:
### What happened

I have multiple webcam and I put the [webcam] config in moonraker.conf, I was expecting that in Mainsail, Fluidd or everything else I will see the webcams already populated.
This is not the case, I don't understand also how it works the camera config in DB. I mean I can insert the webcam config in Mainsail but this is not reflected in Fluidd, and worst part not even in the Timelapse component that it seems to load a "Default" camera on /webcam endpoint instead of using the one selected in Mainsail.

### Client

Mainsail, Fluidd, MobileRaker

### Browser

Chrome

### How to reproduce

Write webcam config
Nothing is loaded at restart

Arksine, 2023-06-10:
Moonraker provides APIs to access webcam data in the configuration, however it is up to each front end to implement them.  As far as I can tell the latest release of Fluidd has support for them.  It doesn't appear that Mainsail does.  I can't speak to Mobileraker or the Timelapse extension, however it sounds like they need to add support.

Arksine, 2023-07-03:
Discussion seems to have concluded on this issue, so I am closing it.

closed 2023-07-03 by Arksine