Arksine/moonraker #674: Bump lmdb to ver 1.4.1 prevents moonraker startup

bavistock, 2023-06-20:
### What happened

Moonraker service won't start, message in logs:

Jun 20 09:13:03 klipper python[6807]: Traceback (most recent call last):
Jun 20 09:13:03 klipper python[6807]:   File "/home/pi/moonraker/moonraker/server.py", line 271, in load_component
Jun 20 09:13:03 klipper python[6807]:     component = load_func(config)
Jun 20 09:13:03 klipper python[6807]:   File "/home/pi/moonraker/moonraker/components/database.py", line 941, in load_component
Jun 20 09:13:03 klipper python[6807]:     return MoonrakerDatabase(config)
Jun 20 09:13:03 klipper python[6807]:   File "/home/pi/moonraker/moonraker/components/database.py", line 100, in __init__
Jun 20 09:13:03 klipper python[6807]:     self.lmdb_env = lmdb.open(self.database_path, map_size=MAX_DB_SIZE,
**Jun 20 09:13:03 klipper python[6807]: AttributeError: module 'lmdb' has no attribute 'open'**
Jun 20 09:13:03 klipper python[6807]: [server.py:main()] - Moonraker Error
Jun 20 09:13:03 klipper python[6807]: Traceback (most recent call last):
Jun 20 09:13:03 klipper python[6807]:   File "/home/pi/moonraker/moonraker/server.py", line 547, in main
Jun 20 09:13:03 klipper python[6807]:     server.load_components()
Jun 20 09:13:03 klipper python[6807]:   File "/home/pi/moonraker/moonraker/server.py", line 236, in load_components
Jun 20 09:13:03 klipper python[6807]:     self.load_component(config, component)
Jun 20 09:13:03 klipper python[6807]:   File "/home/pi/moonraker/moonraker/server.py", line 271, in load_component
Jun 20 09:13:03 klipper python[6807]:     component = load_func(config)
Jun 20 09:13:03 klipper python[6807]:   File "/home/pi/moonraker/moonraker/components/database.py", line 941, in load_component
Jun 20 09:13:03 klipper python[6807]:     return MoonrakerDatabase(config)
Jun 20 09:13:03 klipper python[6807]:   File "/home/pi/moonraker/moonraker/components/database.py", line 100, in __init__
Jun 20 09:13:03 klipper python[6807]:     self.lmdb_env = lmdb.open(self.database_path, map_size=MAX_DB_SIZE,
Jun 20 09:13:03 klipper python[6807]: AttributeError: module 'lmdb' has no attribute 'open'
Jun 20 09:13:03 klipper python[6807]: [server.py:main()] - Server Shutdown
Jun 20 09:13:03 klipper systemd[1]: moonraker.service: Main process exited, code=exited, status=1/FAILURE
Jun 20 09:13:03 klipper systemd[1]: moonraker.service: Failed with result 'exit-code'.

### Client

Mainsail

### Browser

Other or N/A

### How to reproduce

upgraded to latest version of moonraker yesterday, broken since then

### Additional information

reverting to lmdb ver 1.3.0 allows moonraker service to start

bavistock, 2023-06-20:
re-upgraded lmdb to ver 1.4.1 and issue is resolved

closed 2023-06-20 by bavistock

jeffyl, 2023-06-24:
I just had the same problem, and rebuilding the virtualenv with `~/moonraker/scripts/install-moonraker.sh -r` worked in case anyone else runs into this problem.