Arksine/moonraker #681: Unable to upload files from slicers

Myklt8, 2023-07-07:
### What happened

Cura and Orca Slicers do not upload to moonraker, they time out
I reimaged my laptop that runs Klipper, updated the IP address, and added the port and they still won't upload
Times out after a very long time. I am seeing this a lot on FB. 

### Client

Mainsail

### Browser

Firefox

### How to reproduce

Just slice any file and try to upload

### Additional information

[moonraker.log](https://github.com/Arksine/moonraker/files/11984664/moonraker.log)

Arksine, 2023-07-07:
According to the log Moonraker isn't receiving the request.  Based the description of your issue it sounds like your laptop has port 7125 firewalled.   You'll need to open up the port if you want to access Moonraker directly from another machine on your network.

If you have nginx or another reverse proxy running you can likely test this by removing the port in your slicer.  The thing to keep in mind with nginx is that you need to make sure the nginx timeout isn't too short, as per #556.

Myklt8, 2023-07-07:
That was it, Thank you!!!!
   Michael Tate

Arksine, 2023-07-09:
Good to hear, thanks.

closed 2023-07-09 by Arksine

Myklt8, 2023-07-09:
Actually started again, even with the port open Tried different slicers with the same result 

Closed #681 as completed.

Myklt8, 2023-07-09:
I disabled the firewall completely and all is good again