Arksine/moonraker #683: section server is not a valid config section/uds error

Serox13, 2023-07-09:
### What happened

HI
I have had trouble with moonraker since the start of my klipper install. If i include the [moonraker.conf] i have constant errors
the 2 i frequently have and no matter what i do doesn't seem to fix it. I have followed the guides to no avail
I add the [server] section and it always throw back the error is not a valid config section. i also get a uds error no matter what address i use for it. I also get the not a valid config section on [timelapse] as well. any help would be greatful
Thanks

here is my moonraker.conf file
[server]
host: 0.0.0.0
port: 7125
ssl_port: 7130

klippy_uds_address: /home/pi/printer_data/comms/klippy.sock
#   Default is /tmp/klippy_uds.
max_upload_size: 4096
#   The maximum size allowed for a file upload (in MiB).  Default is 1024 MiB.
max_websocket_connections:50
#   The default is 50.
[data_store]
temperature_store_size: 600
gcode_store_size: 5000

[authorization]
enable_api_key: True
#The default is True.
login_timeout:99
max_login_attempts:5
trusted_clients = 
	10.0.0.0/8
	127.0.0.0/8
	169.254.0.0/16
	172.16.0.0/12
	192.168.0.0/16
	FE80::/10
	::1/128
cors_domains:
  http://klipper-printer.local
  http://second-printer.local:7125
#   Enables CORS for the specified domains.  One may specify * if they wish
#   to allow all domains, however this should be an option reserved for
#   client developers and not used in production.  A * can also be used
#   to specify a wildcard that matches several domains.  For example:
#     *.local
#     http://*.my-domain.com
#     *.my-domain.com:*
#   Are all valid entries.  However, a wildcard may not be specified in
#   the top level domain:
#      http://my-printer.*
#   The above example will be rejected.
#   When CORS is enabled by adding an entry to this option, all origins
#   matching the "trusted_clients" option will have CORS headers set as
#   well.  If this option is not specified then CORS is disabled.
force_logins: False
default_source: moonraker

[history]

[update_manager]
channel: dev
refresh_interval: 168

[update_manager mainsail-config]
type: git_repo
primary_branch: master
path: ~/mainsail-config
origin: https://github.com/mainsail-crew/mainsail-config.git
managed_services: klipper

[update_manager mainsail]
type: web
channel: stable
repo: mainsail-crew/mainsail
path: ~/mainsail

# Crowsnest update_manager entry
[update_manager crowsnest]
type: git_repo
path: ~/crowsnest
origin: https://github.com/mainsail-crew/crowsnest.git
managed_services: crowsnest
install_script: tools/pkglist.sh

### Client

Mainsail

### Browser

Chrome

### How to reproduce

not sure how to reproduce the error

### Additional information

[moonraker (4).log](https://github.com/Arksine/moonraker/files/11995514/moonraker.4.log)

Arksine, 2023-07-09:
If you are new to Klipper I can see how this might be confusing, I will do my best to explain it.  Klipper and Moonraker are separate applications, each with their own configuration files.   By default the primary configuration file for Klipper is named `printer.cfg` and the primary configuration file for Moonraker is `moonraker.conf`.  Both applications support including additional configuration files.  Typically configuration files for Klipper should have a `.cfg` extension, and configuration files for Moonraker will have `.conf` section.

The TLDR is that the `[server]` section is valid for Moonraker, not Klipper.  Do not inlcude `moonraker.conf` in `printer.cfg` or any other Klipper configuration file (and likewise, do not include Klipper configuration files in `moonraker.conf`).  Timelapse is an unofficial extension for Moonraker and should not be included in any Klipper configuration file.

Serox13, 2023-07-09:
Understood. Thank you so much for explaining it to me and helping with this issue. Muchly appreciated

Arksine, 2023-07-22:
You're welcome.  I'm going to go ahead and close this as it seems like you have things under control.

closed 2023-07-22 by Arksine