Arksine/moonraker #684: Metadata not available for <>

plugowski, 2023-07-14:
### What happened

After uploading file directly from PrusaSlicer, file cannot be properly paresd, what causes issue with displaying thumbnails in fluid.

moonraker log shows that error after each upload:

`[ERROR] 2023-07-14 11:46:58,775 [root] [common:build_error:575] JSON-RPC Request Error: -32601
{"code":"key86", "msg": "Metadata not available for <misc/CORPO_45m_0.10mm_190C_PLA_ENDER3S1PRO.gcode>", "values": ["misc/CORPO_45m_0.10mm_190C_PLA_ENDER3S1PRO.gcode"]}
Traceback (most recent call last):
  File "/usr/share/moonraker/moonraker/common.py", line 541, in execute_method
    result = await callback(params)
  File "/usr/share/moonraker/moonraker/websockets.py", line 115, in func
    ip_addr=sc.ip_addr, user=sc.user_info))
  File "/usr/share/moonraker/moonraker/components/file_manager/file_manager.py", line 417, in _handle_metadata_request
    """{"code":"key86", "msg": "Metadata not available for <""" + requested_file + """>", "values": [""" + '"' + requested_file + '"' + """]}""", 404)
moonraker.utils.ServerError: {"code":"key86", "msg": "Metadata not available for <misc/CORPO_45m_0.10mm_190C_PLA_ENDER3S1PRO.gcode>", "values": ["misc/CORPO_45m_0.10mm_190C_PLA_ENDER3S1PRO.gcode"]}
`

### Client

Fluidd

### Browser

Chrome

### How to reproduce

Upload file from PrsaSlicer.

Arksine, 2023-07-14:
The traceback isn't enough information to determine the issue, it is necessary to attach the full `moonraker.log` reproducing the error.

plugowski, 2023-07-14:
As Creality claims on their formware page:

`2. Klipper is updated to the open source version V0.11.0, and moonraker (v0.8.0), fluidd (V1.24.0), mainsail (V2.5.1) are updated to the latest version`

So they used latest version of moonraker, and not specially modified.

Arksine, 2023-07-14:
According to [this repo](https://github.com/CrealityOfficial/Creality_Sonic_Pad) the Sonic Pad is using a [fork](https://github.com/CrealityTech/moonraker_ys/tree/65167c103d65763eef0d4fee5edc6a8cb43ecea5).

plugowski, 2023-07-18:
I understand, but anyway is there anything which I can even manually fix in parser?
That's my full log since printer start.

```
--------------------Moonraker Log Start--------------------
data_path: /mnt/UDISK/.moonraker_database
is_default_data_path: False
config_file: /mnt/UDISK//printer_config3/moonraker.conf
startup_warnings: []
verbose: False
debug: False
asyncio_debug: False
is_backup_config: False
is_python_package: False
log_file: /mnt/UDISK/printer_logs3/moonraker.log
software_version: ?
python_version: 3.7.8 (default, Jun 06 2023, 09:42:28)  [GCC 6.4.1]
[INFO] 2023-07-18 07:53:06,395 [root] [loghelper:__init__:79] data_path: /mnt/UDISK/.moonraker_database
[INFO] 2023-07-18 07:53:06,395 [root] [loghelper:__init__:79] is_default_data_path: False
[INFO] 2023-07-18 07:53:06,395 [root] [loghelper:__init__:79] config_file: /mnt/UDISK//printer_config3/moonraker.conf
[INFO] 2023-07-18 07:53:06,396 [root] [loghelper:__init__:79] startup_warnings: []
[INFO] 2023-07-18 07:53:06,396 [root] [loghelper:__init__:79] verbose: False
[INFO] 2023-07-18 07:53:06,396 [root] [loghelper:__init__:79] debug: False
[INFO] 2023-07-18 07:53:06,398 [root] [loghelper:__init__:79] asyncio_debug: False
[INFO] 2023-07-18 07:53:06,398 [root] [loghelper:__init__:79] is_backup_config: False
[INFO] 2023-07-18 07:53:06,399 [root] [loghelper:__init__:79] is_python_package: False
[INFO] 2023-07-18 07:53:06,399 [root] [loghelper:__init__:79] log_file: /mnt/UDISK/printer_logs3/moonraker.log
[INFO] 2023-07-18 07:53:06,399 [root] [loghelper:__init__:79] software_version: ?
[INFO] 2023-07-18 07:53:06,399 [root] [loghelper:__init__:79] python_version: 3.7.8 (default, Jun 06 2023, 09:42:28)  [GCC 6.4.1]
[INFO] 2023-07-18 07:53:06,450 [root] [confighelper:read_file:1064] Configuration File '/mnt/UDISK/printer_config3/moonraker.conf' parsed, total size: 1654 B
[INFO] 2023-07-18 07:53:06,451 [root] [server:add_log_rollover_item:202]
#################### Moonraker Configuration ####################

[server]
host = 0.0.0.0
port = 7127
klippy_uds_address = /tmp/klippy_uds3

[file_manager]
queue_gcode_uploads = False
enable_object_processing = True

[data_store]
temperature_store_size = 600
gcode_store_size = 1000

[machine]
provider = none
validate_service = False
validate_config = False

[authorization]
force_logins = True
cors_domains =
	*.local
	*.lan
	*://app.fluidd.xyz
trusted_clients =
	10.0.0.0/8
	127.0.0.0/8
	169.254.0.0/16
	172.16.0.0/12
	192.168.0.0/16
	FE80::/10
	::1/128

[octoprint_compat]

[history]

[update_manager]
enable_auto_refresh = False

[update_manager client fluidd]
type = web
repo = cadriel/fluidd
path = ~/fluidd

[timelapse]
output_path = /mnt/UDISK/.crealityprint/video
frame_path = /mnt/UDISK/.crealityprint/snapshot/
ffmpeg_binary_path = /usr/bin/ffmpeg

#################################################################
All Configuration Files:
/mnt/UDISK/printer_config3/moonraker.conf
#################################################################
[ERROR] 2023-07-18 07:53:07,755 [root] [database:__init__:120] tianbao: ##########/mnt/UDISK/.moonraker_database
[ERROR] 2023-07-18 08:19:46,969 [root] [common:build_error:575] JSON-RPC Request Error: -32601
{"code":"key86", "msg": "Metadata not available for <printer/RPI-TOP-25FAN_2h1m_0.20mm_190C_PLA_ENDER3S1PRO.gcode>", "values": ["printer/RPI-TOP-25FAN_2h1m_0.20mm_190C_PLA_ENDER3S1PRO.gcode"]}
Traceback (most recent call last):
  File "/usr/share/moonraker/moonraker/common.py", line 541, in execute_method
    result = await callback(params)
  File "/usr/share/moonraker/moonraker/websockets.py", line 115, in func
    ip_addr=sc.ip_addr, user=sc.user_info))
  File "/usr/share/moonraker/moonraker/components/file_manager/file_manager.py", line 417, in _handle_metadata_request
    """{"code":"key86", "msg": "Metadata not available for <""" + requested_file + """>", "values": [""" + '"' + requested_file + '"' + """]}""", 404)
moonraker.utils.ServerError: {"code":"key86", "msg": "Metadata not available for <printer/RPI-TOP-25FAN_2h1m_0.20mm_190C_PLA_ENDER3S1PRO.gcode>", "values": ["printer/RPI-TOP-25FAN_2h1m_0.20mm_190C_PLA_ENDER3S1PRO.gcode"]}
[ERROR] 2023-07-18 08:20:37,931 [root] [common:build_error:575] JSON-RPC Request Error: -32601
{"code":"key86", "msg": "Metadata not available for <printer/RPI-TOP-25FAN_2h1m_0.20mm_190C_PLA_ENDER3S1PRO.gcode>", "values": ["printer/RPI-TOP-25FAN_2h1m_0.20mm_190C_PLA_ENDER3S1PRO.gcode"]}
Traceback (most recent call last):
  File "/usr/share/moonraker/moonraker/common.py", line 541, in execute_method
    result = await callback(params)
  File "/usr/share/moonraker/moonraker/websockets.py", line 115, in func
    ip_addr=sc.ip_addr, user=sc.user_info))
  File "/usr/share/moonraker/moonraker/components/file_manager/file_manager.py", line 417, in _handle_metadata_request
    """{"code":"key86", "msg": "Metadata not available for <""" + requested_file + """>", "values": [""" + '"' + requested_file + '"' + """]}""", 404)
moonraker.utils.ServerError: {"code":"key86", "msg": "Metadata not available for <printer/RPI-TOP-25FAN_2h1m_0.20mm_190C_PLA_ENDER3S1PRO.gcode>", "values": ["printer/RPI-TOP-25FAN_2h1m_0.20mm_190C_PLA_ENDER3S1PRO.gcode"]}
```

Arksine, 2023-10-27:
I hope that the Sonic Pad has since resolved this in their distribution.  Unfortunately there is little I can recommend, however there may be people either on the Discourse or Discord that have experience with this device and can provide assistance.

closed 2023-10-27 by Arksine