Arksine/moonraker #685: ./fetch-apikey.sh causes lmdb.ReadonlyError: mdb_dbi_open: Permission denied

nicolas-schreiber, 2023-07-14:
### What happened

```
pi@octopi:~/moonraker/scripts $ ./fetch-apikey.sh 
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/pi/moonraker-env/lib/python3.9/site-packages/lmdb/__main__.py", line 24, in <module>
    lmdb.tool.main()
  File "/home/pi/moonraker-env/lib/python3.9/site-packages/lmdb/tool.py", line 617, in main
    DB = ENV.open_db(_to_bytes(opts.db))
lmdb.ReadonlyError: mdb_dbi_open: Permission denied
```

### Client

Other

### Browser

Other or N/A

### How to reproduce

Simply installing moonraker after the guide and then calling `./fetch-apikey.sh`

Arksine, 2023-07-14:
That is not normal.  Can you attach `moonraker.log`?  It would be useful to see if Moonraker itself can't open the database file.

Some additional things you can try:
1) Restarting Moonraker to see if closing the DB resolves the issue.
2) Using curl:
```
curl http://127.0.0.1:7125/access/api_key
```
Note the latter will only work if `127.0.0.1` is added to `trusted_clients` option in the `[authorization]` configuration section.

Arksine, 2024-02-04:
As no one has been able to provide further feedback I'm going to close this.  If you are still having issues I recommend asking for assistance on the Klipper Discourse.

closed 2024-02-04 by Arksine