Arksine/moonraker #686: Update moonraker kills webcam - Logitech C920

CD1010, 2023-07-16:
### What happened

So my web cam stopped working and i reinstalled from raspi pi imager.  I installed mainsail 32 bit image.  Camera came back.  Then i updated moonraker, just that. Camera died again.  i use a Logitech C920.

I have never had trouble with that camera.

Steps:  Make image  for Mainsail 32 bit using RaspiPi imager 
2) restore printer.cfg etc.
3) camera works. 
4) Update nothing but moonraker.
5) camera dies.

### Client

Mainsail

### Browser

Chrome

### How to reproduce

Steps:  Make image  for Mainsail 32 bit using RaspiPi imager 
2) restore printer.cfg etc.
3) camera works. 
4) Update nothing but moonraker.
5) camera dies.

Arksine, 2023-07-22:
You need to attach `moonraker.log` for me to look into this.  Additionally, please clarify what you mean by the "camera dies".  Do you mean that its no longer available in Mainsail?  Not enumerating on the bus?  Is the stream accessible if you enter the stream url directly in the browser?

Moonraker does not in any way interact directly with the camera.  It can store configuration for the camera, and the way the configuration is stored may have changed between the installed version and the updated version.  In this case, you should be able to get the camera back by removing and adding it again in Mainsail.

CD1010, 2023-07-23:
Sorry, I wiped and reinstalled.  Camera stream went black once I update moonraker,  so I don't now. I tried adding and removing camera, did not help.
Kpippe4 firmware not latest. Could that matter.

tafypz, 2023-07-24:
I had a similar issue after updating moonraker through mainsail.
running make clean, make build, make install on moonraker solved it for me.

Arksine, 2023-10-27:
As the discussion on this issue seems to have concluded I am going to close this.  Thanks.

closed 2023-10-27 by Arksine