Arksine/moonraker #691: Moonraker update not working with 3rd party repo/branches correctly - wipes data and links back to main/master.

tinymachines3d, 2023-07-26:
### What happened

Shows invalid with a custom repo/ branch that is not main/master which was working fine prior to this update. 

Soft recovery would clear and allow update then immediately show invalid again. 

Hard recover resets the git branch to master and wipes the original data completely. 

### Client

Mainsail, Other

### Browser

Chrome

### How to reproduce

Not sure

### Additional information

[image]

Arksine, 2023-07-26:
I will need you to attach log files that reproduce and show this behavior to diagnose what exactly is going on.   The first image shows a repo that is marked as invalid because it is dirty, not because of a branch mismatch.  A branch mismatch should not invalidate a repo, however there was a brief period of time where the newest version of Mainsail would show it as a warning.  What isn't clear to me is if the repo is showing as dirty after the update or showing the aforementioned warning.  It also isn't clear to me how the repo in the screenshot got modified to begin with, is there some application or script modifying this repo?

FWIW, a soft recovery should fix a dirty repo.  A hard recovery re-clone's the repo, you will end up on whatever the default branch of the repo is.  Moonraker doesn't switch the branch after a clone, it must be done via the CLI.

Arksine, 2023-07-26:
As an addendum I just did the following on the latest Moonraker commit:

- Cloned your `klipper_elements` repo
- Switched to the `troodon_2.0` branch
- Reset the HEAD back 3 commits so an available update will be detected
-  Added the following configuration to my test instance of Moonraker:
```
[update_manager elements]
type: git_repo
channel: dev
primary_branch: troodon_2.0
origin: https://github.com/tinymachines3d/klipper_elements.git
path: /home/eric/klipper_elements
is_system_service: False
```

I then used Moonraker to update the repo via mainsail and it worked as expected.

Arksine, 2024-02-04:
As no one has been able to provide further feedback I'm going to close this.  If you are still having issues I recommend asking for assistance on the Klipper Discourse.

closed 2024-02-04 by Arksine

rosenrot00, 2025-01-08:
I have to re-open this one. I just experienced something that wasn't as expected - at least to me. To me this is related to the OP's title.

[update_manager test]
type: git_repo
channel: dev
primary_branch: dev

What I did is:
1. Clone to git repo using a terminal command. This results to being in the master branch.
2. When hitting update in Moonraker, the branch stays on master even though the primary_branch was set to dev in Moonraker

I expect Moonraker to change the branch to dev and check for updates. I would like Moonraker to perform catch updates on the specified branch which it currently doesn't. I expect that to happen because otherwise the primary_branch setting is useless, isn't it?