Arksine/moonraker #693: Tasmota Configuration

stefan63, 2023-07-27:
### What happened

adress Tasmota1  Do not work (SSID/Hostname)
adress: 102.168.1.11  WORKS!

### Client

Fluidd

### Browser

Chrome

### How to reproduce

Simply try to enter Hostname of Tasmota device insted of IP and se if it works.

### Additional information

It did work in the past but I can not remember in wich version :(

stefan63, 2023-07-28:
> :
> It is necessary to reproduce the issue and attach moonraker.log for me to
> provide meaningful feedback. I suspect this is a local DNS issue, as I
> can't reproduce it on my tasmota devices.

Thank you for your answer I will look in to it, and I bet you are right it
something with my local DNS.
Love your work, kind regards/Stefan

Den tors 27 juli 2023 kl 18:08 skrev Eric Callahan ***@***.***

Arksine, 2023-10-27:
As this discussion seems to have concluded I am going to close this.  Thanks.

closed 2023-10-27 by Arksine