Arksine/moonraker #694: Webcam API: Moonraker does not validate flipX and flipY

Clon1998, 2023-07-28:
### What happened

Several users reported that they are having trouble using the webcam on Mobileraker. After conducting an investigation, it seems that the issue may lie with Mainsail, which recently switched to the Webcam API. Specifically, when a user edits a webcam through Mainsail and Mainsail writes it back to Moonraker, Moonraker fails to validate certain fields, such as filpX and flipY. As a result, Mainsail may write an integer instead of a boolean to the field. 

I have already reported this [issue](https://github.com/mainsail-crew/mainsail/issues/1493) to the Mainsail repository. However, I believe it would be wise for Moonraker to validate the values in the future to avoid any potential problems with the webcam API for other UIs.

### Client

Mainsail, Fluidd, KlipperScreen, MobileRaker

### Browser

Other or N/A

### How to reproduce

Edit a webcam via mainsail and via fluidd/moonraker.
Mainsail will write flip values as ints.
Fluidd/Mobileraker as booleans, as documented in the examples and also probably expected by looking into the [code](https://github.com/Arksine/moonraker/blob/937c766ded8213d43ff58bde5355e31e13e35b88/moonraker/components/webcam.py#L412-L413).

meteyou, 2023-07-28:
flipX/flipY is an outdated attribute. The new attribute names are flip_horizontal and flip_vertical. Pls check the Moonraker docs for all attribute names: https://moonraker.readthedocs.io/en/latest/web_api/#webcam-apis

Arksine, 2023-07-28:
Meteyou is correct.  flipX and flipY were database values set by mainsail in the original database.  When I set the APIs I decided to change the names to snake case (to be consistent with other parameters in Moonraker), and renamed these two to `horizontal` and `vertical` respectively.

Arksine, 2023-08-25:
I believe that this is resolved so I am going to close this.  Thanks.

closed 2023-08-25 by Arksine