Arksine/moonraker #695: Enable support for OpenRC and other init systems

louisrousseau, 2023-07-31:
### Is your feature request related to a problem? Please describe

I've recently managed to install Klipper, Moonraker and Mainsail on an old phone using postmarketOS - an Alpine Linux derived distro - not Android.

Main advantages are that this already supports several old phones with a much more minimalist busybox/Linux install than Android and uses an up-to-date mainline kernel.

One problem I've had so far is that some parts of Moonraker expect a working `systemd` initialization system to be present, e.g. I can't reboot the host via Moonraker from Mainsail and have to do so via ssh instead.

### Describe the solution you'd like

Some way to override those commands that depend on systemd.

nelsongraca, 2024-03-09:
would a solution where you define the commands used to start/stop/restart reboot and shutdown the services work for you? I also run distros with OpenRC and my klipper/mainsail etc runs in docker so custom commands would solve my problem and it would be useful to me too.
maybe a custom provider in `machine` section along with more config for the commands?

louisrousseau, 2024-03-09:
It would probably have though an update eventually broke my postmarketOS, by that time Raspberry Pi 4s had come back in stock so I moved back to a much more "standard" Raspberry Pi OS installation.

nelsongraca, 2024-03-09:
alright, might still implement it as it will still be useful to me.

ChasonDeshotel, 2025-01-23:
I'm running on Alpine, which is OpenRC. This would be useful to me as well