Arksine/moonraker #701: Shelly power device - special character `*` in password is not working

MoJo2600, 2023-08-15:
### What happened

My printer is connected to a shelly 1 to power up the printer from the GUI. It was working, but after an update (I don't know the version) the device stopped working. I investigated the issue in version v0.8.0-136 and the logs show an error `401 Unauthorized` when initializing the shelly device. I tried moving the password to a `moonraker.secrets` file. But the result was the same. Only after I removed the special character `*` from the password and the password only containing characters and numbers, the device started working again. I did not try other special characters or other devices. I believe the issue is with the jinja templating of the password string.

### Client

Mainsail

### Browser

Chrome

### How to reproduce

Set the password of a shelly device to e.g. `AbCdEfG*123HiJ` the device will not work and the error in the logs show `401 Unauthorized`

### Additional information

```
-------------------- Log Start | Mon Aug 14 10:46:51 2023 --------------------
data_path: /home/klipper/printer_data
is_default_data_path: False
config_file: /home/klipper/klipper_config/moonraker.conf
startup_warnings: []
verbose: False
debug: False
asyncio_debug: False
is_backup_config: False
is_python_package: False
instance_uuid: 28fee33943d4485fa2c4886c763423dd
unix_socket_path: /home/klipper/printer_data/comms/moonraker.sock
software_version: v0.8.0-136-g797c623
git_branch: master
git_remote: origin
git_repo_url: https://github.com/Arksine/moonraker.git
modified_files: []
unofficial_components: []
log_file: /home/klipper/klipper_logs/moonraker.log
python_version: 3.9.2 (default, Mar 12 2021, 04:06:34)  [GCC 10.2.1 20210110]

#################### Moonraker Configuration ####################

[server]
host = 0.0.0.0
port = 7125
klippy_uds_address = /tmp/klippy_uds

[database]

[authorization]
trusted_clients = 
	10.0.0.0/8
	127.0.0.0/8
	169.254.0.0/16
	172.16.0.0/12
	192.168.0.0/16
	FE80::/10
	::1/128
cors_domains = 
	*.lan
	*.local
	*://my.mainsail.xyz
	*://app.fluidd.xyz

[file_manager]

[octoprint_compat]

[history]

[... 460 lines cut ...]
leaf_pages: 1
overflow_pages: 0
entries: 7
timelapse:
psize: 4096
depth: 1
branch_pages: 0
leaf_pages: 1
overflow_pages: 0
entries: 1
update_manager:
psize: 4096
depth: 2
branch_pages: 1
leaf_pages: 2
overflow_pages: 1
entries: 7
webcams:
psize: 4096
depth: 1
branch_pages: 0
leaf_pages: 1
overflow_pages: 0
entries: 2
```

Barsonax, 2024-08-10:
Seems to happen with other special characters like ^ and $ as well

Arksine, 2024-08-11:
Hi.  I have investigated this and believe that I have corrected the issue in commit e66744b6e9b3475cb2fb3908777112992a01bf76.  That said, I do not have a Shelly device, so I cannot test it.

Barsonax, 2024-08-11:
Will it help if I test it and if so how or will you just release it?

Arksine, 2024-08-11:
I pushed the commit as I don't think it will cause a regression.  You can update Moonraker and test to see if it resolves your issue.

Barsonax, 2024-08-11:
I updated to v0.9.2-5-ge66744b6 and this fixes the problem for me

MoJo2600, 2024-08-16:
I can confirm it is working now with moonraker `v0.9.2-6-gf735c041`

closed 2024-08-16 by MoJo2600