Arksine/moonraker #707: Notifier sends notification during a timelapse pause event

mpaw, 2023-08-25:
### What happened

When `moonraker-timelapse` parks the print head, it has to use the `PAUSE` macro. This causes the `notifier` to send a notification on every single layer change. This [comment](https://github.com/mainsail-crew/moonraker-timelapse/issues/42#issuecomment-1045006129) mentions how the notifier should detect it is a pause from the timelapse and not trigger a notification.

### Client

Fluidd, KlipperScreen

### Browser

Chrome, Brave

### How to reproduce

Set up a notifier for the `PAUSE` event, turn on timelapse with the layer macro and start a print. You will get a notification on every layer.

### Additional information

Not needed

Arksine, 2023-08-25:
From the described behavior and linked issue, it appears that the timelapse macros are introducing this behavior.  This isn't a bug in Moonraker, as its correctly reacting to the pause.  It isn't feasible for Moonraker to filter events based on behavior introduced by custom macros, as they are subject to change and would be impossible to support.

That said, it is likely that this can be accounted for by configuring your notification in `moonraker.conf` to omit the `paused` event.  You can then add a new notifier that reacts to a `gcode` event and override your `PAUSE` macro in `printer.cfg` to fire that event when the timelapse macro reports `is_paused == False`.

mpaw, 2023-08-25:
@Arksine I agree with what you said but it seems like the only way that `moonraker-timelapse` can park the print head is by triggering a `PAUSE`. Should there be a separate `PAUSE` event that could be used in situations like these that could be ignored by the notifier? I definitely want to get notifications when I run out of filament or there is a layer change, but not on every layer when a timelapse picture is taken.

As a workaround for now I'll create a custom macro notifier for when I run out of filament and on filament changes in the slicer. Just throwing it out there that maybe there could be a better solution so people in the future don't have to create custom GCODE macros to get around this behavior.

Arksine, 2023-08-25:
The long term solution is for the timelapse plugin to implement its Klipper side logic using something other than gcode macros.   It could probably benefit from the solution proposed [here](https://klipper.discourse.group/t/possible-klipper-plugins-instead-of-macros/9819/15), as the gcode handlers could probably be implemented in the timelapse component.  There would be no need to issue a "pause",  the handlers could simply block execution until they detect that rendering is complete.