Arksine/moonraker #710: Moonraker Warning: Moonraker does not have permission to access path

Kiwii0411, 2023-09-04:
### What happened

Hi, im getting a Moonraker warning :

Moonraker does not have permission to access path (/home/user/printer_data/config) for (config).

[image]

[image]

I tried using the script that i found on https://github.com/Arksine/moonraker/issues/516 ./scripts/data-path-fix.sh but i get an error saying :

Error: unable to find config: /home/user/klipper_config/moonraker.conf

Can somebody please help, im stuck

EditIm using Klipper with Moonraker and Mainsail on an old laptop with Linux mint on it connected to a Sovol Sv06 and everything was installed using kiauh

### Client

Mainsail

### Browser

Chrome

### How to reproduce

I dont have a clue

Arksine, 2023-10-15:
I'm unsure if you were able to find assistance for this, but it looks like there is an issue with read/write access to the config directory.  I would personally recommend Ubuntu over Linux Mint if the PC is exclusively intended to host Klipper.

I would recommend visiting one of the discords for generic assistance, as there are likely going to be people who can assist you with more generic issues such as this one.

Arksine, 2024-02-04:
As no one has been able to provide further feedback I'm going to close this.  If you are still having issues I recommend asking for assistance on the Klipper Discourse.

closed 2024-02-04 by Arksine