Arksine/moonraker #711: Support configuring a context path that all endpoints are under.

Laikulo, 2023-09-07:
### Is your feature request related to a problem? Please describe

No.

### Describe the solution you'd like

The ability to configure a prefix (e.g. /printer1, /moonraker/, /api) that is prepended to all routes, to allow for multiple moonrakers to be easily accessed through a single reverse proxy.

This would allow for something like the following from the user perspective

printer.example.net/ -> Welcome page, or redirect to web based frontend
    /moonraker/printer1 -> API for printer1
        /server -> current behavior for /server.
        [more]
    /moonraker/printer2 -> API for printer 2
        [see above]
    /cockpit -> Cockpit-ws (server management UI)
    /mainsail -> Static server serving mainsail application
    /auth -> System-specific authentication system. Possibly issues cookies that result in X-Auth-User headers to downstreams.

This first level routing would be handled by a revers proxy, such as Caddy, nginx, envoy, or similar.

### Describe alternatives you've considered

Using the reverse proxy to rewrite requests on the wire, so that they arrive at moonraker with shortened paths.

Arksine, 2023-09-12:
I'm not opposed to this idea, however I don't think it will be useful without support from the frontends.  Currently none of them can access Moonraker's APIs unless they are on the root path, so they will never request the path with the prefix prepended.  

Do you have another use case for this now?  Or is this something you plan on requesting from the front ends?  As you mentioned, they could add support for a prefix now, then use the reverse proxy to rewrite the API and direct it to the correct instance.

Laikulo, 2023-09-12:
My main use case for this is to allow hosts running multiple instances to expose them over a single HTTP(S) Port, where name-based virtual hosting is not available, however, one of my other schemes involves running a series of printers with multiple small hosts each running a Moonraker instance behind a single reverse proxy.

I was not aware that the front ends didn't actually support this, as most of the ones I've tried will accept paths. If I can find some time over the next few days, I'll set up a rewriting reverse proxy entry, and report back.

Laikulo, 2023-09-12:
The main reason for wanting to avoid a rewrite-based solution, is that if Moonraker generates any URLs as part of its responses, those won't be rewritten. Reverse proxies often have special handling for redirects, but not much more. Come to think of it, I don't think Moonraker does generate such responses...

Laikulo, 2023-09-13:
Set up a test with the following caddy config

```
voron.haun.haus:80, :80 {
        # Set this path to your site's directory.
        root * /srv/www/htdocs

        reverse_proxy /server/* localhost:7125
        reverse_proxy /access/* localhost:7125
        reverse_proxy /machine/* localhost:7125
        reverse_proxy /api/* localhost:7125
        reverse_proxy /printer/* localhost:7125
        reverse_proxy /websocket localhost:7125
        reverse_proxy /websocket/* localhost:7125

        # Enable the static file server.
        file_server
}

moonrakertest.haun.haus:80 {
        route /moonraker/* {
                uri strip_prefix /moonraker
                reverse_proxy localhost:7125
        }
        root * /srv/www/htdocs
        file_server
}
```

* Fluidd initially accepted the URL, but dispatched requests to / anyways
* Mainsail (with instancedb: browser) accepted the URL, but incorporated the port number into the request. (e.g. ws:/moonrakertest.haun.guru/moonraker/:80/websocket)
* Mobileraker refused to accept any URL with a path
* PrusaSlicer'/SuperSlicer/OrcaSlicer's send gcode (via octoprint compat) accepted the path, and was able to upload a file.

Laikulo, 2023-09-13:
So, if this were to make it into moonraker, I would most likely open requests against the frontends to support this.

Laikulo, 2023-09-13:
[image]

Arksine, 2023-10-05:
This should be resolved with the `route_prefix` added on commit ae1d3b039308a84cbc01092f11b4c7c3b84da138.

closed 2023-10-05 by Arksine