Arksine/moonraker #716: Cannot Passthrough Channel to Apprise Notifier

Bupher, 2023-09-23:
### What happened

I am unable to passthrough a specific channel name to apprise notifier.

I am able to get general notifications to work, messages sent from a specific device to work, but not messages to a channel. The apprise syntax says it should be in the form of `pbul://accesstoken/#channel`.

If I leave the syntax exactly like that, but with my channel name, then I just get the general notification. If I try and escape the #, it just never sends a message. Looking at the moonraker log, I can see the line always just ends right after #, no matter what I try and do to escape that character.

`pbul://{secrets.pushbullet_token.token}` -> just sends a general notification
`pbul://{secrets.pushbullet_token.token}/#swx1` -> just sends a general notification
`pbul://{secrets.pushbullet_token.token}/{'#swx1'}` -> no message sent
`pbul://{secrets.pushbullet_token.token}/{{'#swx1'}}` -> no message sent

### Client

Mainsail

### Browser

Chrome

### How to reproduce

Setup klipper with a notifier set to pushbullet, and try and target a pusbullet notification to a specific channel.

### Additional information

[moonraker (1).log](https://github.com/Arksine/moonraker/files/12705733/moonraker.1.log)

Arksine, 2023-10-05:
I don't think you need to worry about escaping anything, apprise should take care of that.   According to the apprise documentation, if you want to send a notification to the swx1 channel the following should work:

```
pbul://{secrets.pushbullet_token.token}/#swx1
```

Looking at the attached log, I don't see this entry.  The closest I see is this:
```
[notifier pushbullet]
url = pbul://{secrets.pushbullet_token.token}%23swx1
events = gcode
body = {event_message}
attach = http://127.0.0.1/webcam/?action=snapshot
```

The other entries appear to send to a device ID or have incorrect syntax.  The above entry is missing the `/` prior to the channel, and as mentioned its escaping the `#` which shouldn't be required.

Bupher, 2023-10-06:
I've uploaded both a fresh log and the config file active when generating that log. If you look between the two, you will see comments from the config are not shown in the log (see the #notifications header in my config). I tried to use `pbul://{secrets.pushbullet_token.token}/#swx1`, but it treats `#swx1`, does not include it in the log, and does not process it.

Config:
[image]

Log:
[image]

[moonraker (4).log](https://github.com/Arksine/moonraker/files/12827724/moonraker.4.log)

[moonrakerConfig.txt](https://github.com/Arksine/moonraker/files/12827749/moonrakerConfig.txt)

Arksine, 2023-10-06:
Ah, I see.  Indeed its an issue with the configuration parser removing the fragment thinking its an inline comment.  I'm working on a fix and should hopefully have it pushed soon.

Arksine, 2023-10-06:
Commit 432924194959de0e25187d4bb999b9794bec553a should resolve this.  Hopefully it doesn't introduce any regressions, dealing with inline comments in the configuration is traditionally tricky.

Bupher, 2023-10-07:
This seems to have solved it, thank you!

closed 2023-10-15 by Arksine